MarlinFirmware/Marlin #26231: [BUG] On Print Start from SD Card Print Never Starts

XxZorro13xX, 2023-09-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I believe this is possibly what https://github.com/MarlinFirmware/Marlin/issues/13766 had, however, they gave bad details and nothing came of it.  Basically, I power on the system, navigate to the file list from the SD card, and select print file X,  File X never starts, printer does not start heating, instead, the heat bed temp stays 0/0.  So I presume the file is not even being read.  The printer is not frozen and i can stop the print like normal.  All other functions of the printer appear to work.  I can print via USB.  If you need any other details please ask.

I am on 2.0.7 version created by Nick Well, I cannot seem to find anyone else with this issue and have only found one temp fix I found on a Reddit post.  It was sending M503, then M500.  However, on a power cycle, the problem appears again.

### Bug Timeline

Old never caught

### Expected behavior

File to start

### Actual behavior

file never initializes

### Version of Marlin Firmware

2.0.7.2

### Printer model

Anycubic Chiron

### Electronics

Stock

### Add-ons

stock

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[https://www.n-wells.co.uk/3DPrinting/Chiron/Marlin%202.0.7.2%20for%20the%20Anycubic%20Chiron.pdf](url)

thisiskeithb, 2023-09-04:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

XxZorro13xX, 2023-09-05:
After flashing 2.1.x that issue is fixed, but now another one has popped up, when starting probing, the process will begin>home all axis> touch first point once and end.

thisiskeithb, 2023-09-05:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2023-11-19 by github-actions[bot]