MarlinFirmware/Marlin #26232: [BUG] Odd behavior at ABL completion.

gaby64, 2023-09-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On (G29) ABL completion, the return to home has odd behavior, it returns at a random slow speed, sometimes faster and sometimes painfully slow, other odd behavior I have observed is it just sitting there at the ABL end point and purging filament at speeds the extruder cannot handle (this one tends to happen after a few prints, so a restart brings it back to only the slow return home)
I have experienced this issue since all the way back to 2021 marlin builds.

The machine is a Two Trees Sapphire Plus, Robin Nano 1.2.

### Bug Timeline

2021

### Expected behavior

ABL completes and returns home at speed of HOMING_FEEDRATE_MM_M

### Actual behavior

ABL completes and returns home at odd and random slow speeds, occasional sit and purge at extruder stripping rates before start of return to home.

### Steps to Reproduce

This is my start Gcode, this where G29 is called.

```
G28 ;Home
G29;
G0 Z5 ; Lift Z
G0 X0 Y0 F8000 ; Go to corner
G0 Z0.15 F500 ; Lower nozzle
G0 E8 F600;  Extrude to prime nozzle
G0 X20 Y20 F8000;
G0 Z4;
```

### Version of Marlin Firmware

02010300

### Printer model

Two Trees Sapphire Plus

### Electronics

Stock

### Add-ons

BLTOUCH, TMC2209 on XY with serial uart, 0.9deg steppers on XY

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_SapphirePlus.zip](https://github.com/MarlinFirmware/Marlin/files/12517306/Marlin_SapphirePlus.zip)

gaby64, 2023-12-29:
Retesting with latest

gaby64, 2024-03-25:
I had not used my printer in a few months, just witnessed the bug again today. So it was still present in December, I will update to latest and test again.

gaby64, 2024-06-25:
issue is still present

thisiskeithb, 2024-06-25:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code. You will need to port your config/settings to this version.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

gaby64, 2024-09-25:
Sorry, I have not had time to follow the procedure to gather more data, the issue is still present though.
It's not on every home or every level, it happens on the second print usually, power cycling resolves it again but only until the second print again.

gaby64, 2024-12-26:
Running latest and issue is still present.

closed 2025-04-10 by github-actions[bot]