MarlinFirmware/Marlin #26246: [BUG] After #26033 probbing become ~2x slower

eoyilmaz, 2023-09-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After #26033 merged probing become immensly slower (around 1.5x slower for 3x3 mesh and most probably 4x for 5x5). I bisected the change to 3815157214 and reverting it fixes the problem.

### Bug Timeline

Introduced with 381515721481d09c4f30ca8a526e90fc39602fd9

### Expected behavior

I expect the BLTouch to probe faster in high speed mode.

### Actual behavior

Probbing is slow

### Steps to Reproduce

1. Create a mesh

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender3 Pro

### Electronics

SKR Mini E3 v3

### Add-ons

CRTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[My configuration files](https://github.com/MarlinFirmware/Marlin/files/12528956/Archive.zip)

Here is a video demonstrating the effect of 381515721481d09c4f30ca8a526e90fc39602fd9:

https://github.com/MarlinFirmware/Marlin/assets/1786804/e394a565-404d-455e-b003-74d175605fbd

eoyilmaz, 2023-09-07:
I would suggest something like this to be used instead of:

https://github.com/MarlinFirmware/Marlin/blob/11f98adcce51f06a85ef0dd64e6e6b16139a34bd/Marlin/src/module/probe.cpp#L600

```c
  #if ENABLED(BLTOUCH)
    #if ENABLED(MEASURE_BACKLASH_WHEN_PROBING)
      // Ensure the BLTouch is deployed. Does nothing if already deployed.
      if (bltouch.deploy())
        return true; // Deploy in LOW SPEED MODE on every probe action
    #else
      if (!bltouch.high_speed_mode && bltouch.deploy())
        return true; // Deploy in LOW SPEED MODE on every probe action
    #endif
  #endif
```

eoyilmaz, 2023-09-27:
Created a PR...

ellensp, 2023-09-27:
Since a pr has been added, closing the issue.

closed 2023-09-27 by ellensp