MarlinFirmware/Marlin #26247: [FR] Check ZMIN before doing homing moves, and raise Z if ZMIN is triggered

vlsi, 2023-09-06:
### Is your feature request related to a problem? Please describe.

If the nozzle is on the bed, then `G28` might move the nozzle without raising Z first.

In other words:
* If ZMIN is triggered, perform Z clearance before moving X and Y
* If ZMAX is triggered, then Marlin could probably skip "Z bump" as it will likely crash into the Z max

At the same time, if I configure `Z_IDLE_HEIGHT Z_HOME_POS`, then Marlin does raise Z first, however, I believe it could be improved as well.

The current code performs two Z bumps when Z is untrusted:
1. Bump Z before homing XY
2. Start Z, and then Marlin would bump Z to ensure probe clearance

I believe the second bump is not really needed, and just one of them is enough. In other words, if as Marlin bumps Z for safely homing XY, it should remember it has enough clearance for deploying Z probe.

### Describe the feature you want

I would like `G28` to check if `ZMIN` is triggered before doing other moves.

I'm using 2.1.2.1, and the essential bits are:
* COREXY
* `#define Z_IDLE_HEIGHT Z_MAX_POS`
* `FIX_MOUNTED_PROBE`
* Both `G28` and `G28 X` reproduce the issue after startup

After startup, the untrusted Z is 350 which is exactly Z_MAX_POS, and I believe `do_z_clearance` performs "move to Z_MAX_POS" which is a no-op move since the "current position is already 350".
https://github.com/MarlinFirmware/Marlin/blob/09d0b4d15228b3efb6fd765f1e1574f521df2e27/Marlin/src/module/motion.cpp#L785

rondlh, 2023-09-08:
> If the nozzle is on the bed, then `G28` might move the nozzle without raising Z first.

I believe you can use Z_CLEARANCE_FOR_HOMING (configuration.h) to achieve your goal.

vlsi, 2023-09-08:
I have `Z_HOMING_HEIGHT` which is I guess 2.1.2.1 name for `Z_CLEARANCE_FOR_HOMING`

rondlh, 2023-09-08:
> I have `Z_HOMING_HEIGHT 5` which is I guess 2.1.2.1 name for `Z_CLEARANCE_FOR_HOMING`

It's the other way around, you should use Z_CLEARANCE_FOR_HOMING.
If you are on Marlin 2.1.2.1 you cannot use Z_HOMING_HEIGHT anymore.
In G28.cpp you can see that G28 will cause the clearance move (unless G28 only moves Z)

classicrocker883, 2023-09-10:
> It's the other way around

actually, just checked, and it *is* `Z_HOMING_HEIGHT` for 2.1.2.1, it was renamed Clearance for Homing in the current *bugfix*.

unless... if what you say is true then the Configuration.h file is wrong.

classicrocker883, 2023-09-10:
in the current bugfix, this function has `void do_z_clearance() {... 
NOMORE(zdest, Z_MAX_POS);`

how about something like 
```y
NOMORE(zdest, Z_MAX_POS - zclear);
```

vlsi, 2023-09-12:
>how about something like
>NOMORE(zdest, Z_MAX_POS - zclear);

I think it should be good, however, "z_map_pos-zclear" should probably be used only when Z is untrusted. If Z is trusted, then the clearance movement should be limited by z_max_pos. WDYT?

classicrocker883, 2023-09-14:
> > how about something like
> > NOMORE(zdest, Z_MAX_POS - zclear);
> 
> I think it should be good, however, "z_map_pos-zclear" should probably be used only when Z is untrusted. If Z is trusted, then the clearance movement should be limited by z_max_pos. WDYT?

something like `if (!axis_is_trusted(Z_AXIS))`

thinkyhead, 2023-09-23:
You should re-test with the latest `bugfix-2.1.x` because homing and probing have been overhauled.

vlsi, 2023-09-24:
There might be at least two cases, and I believe the first of them is not implemented yet.

1) zmin is triggered before homing begins. In that case, firmware can be sure z must be increased before any movements

2) zmin is not triggered. It could be "bed is already far away from the nozzle", "bed is away, but the nozzle touches the printed part", "bed is at z-max". In these cases, the firmware has no way to tell if moving nozzle is safe, and it can not tell if increasing z would be safe either. I guess the current approach is "increase z just in case" which is fine.

I could test it, however, last time I checked 2.1.x, there was no zmin check (case 1) at the of home procedure