MarlinFirmware/Marlin #26251: [BUG] SPINDLE_LASER_ENA_PIN not active when SPINDLE_LASER_PWM_PIN is sending PWM signal

rowokii, 2023-09-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to setup Marling 2.0.9.7 on MELZI_CREALITY (V 1.1.4) controller board as an laser engraver.

Using, 
#define LASER_FEATURE
#define SPINDLE_LASER_USE_PWM
#define SPINDLE_LASER_ACTIVE_STATE HIGH

I tried to test fired the laser with:
M03 S5.1

for a 2% laser power on.

This did not seem to work, so I observed the control pins with an oscilloscope.

Since LASER_SAFETY_TIMEOUT_MS is set to 1000, I get 1 sec burst of PWM signal on the PWM pin, but the ENA pin only briefly goes high at the end of the PWM pulse train on V. 2.0.9.7.  On the latest bugfix version, I don't see any activity on the ENA pin.

### Bug Timeline

New

### Expected behavior

On
M03 S5.1

I expected ENA pin to go high, followed by 1 sec PWM signal on the PWM pin followed by ENA pin to go low.

### Actual behavior

On
M03 S5.1

I get ENA pin staying low, 1 sec PWM signal on the PWM pin, followed by very brief pulse high and then low on the ENA pin

### Steps to Reproduce

1. Configure Marling 2.0.9.7 with my config files
2. send command M03 S5.1
3. observe the PWM and ENA pins
[Config_PINS.zip](https://github.com/MarlinFirmware/Marlin/files/12564440/Config_PINS.zip)

### Version of Marlin Firmware

2.0.9.7

### Printer model

NA

### Electronics

Melzi_creality (V1.1.4)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-09-09:
you seem to be using

    #define SPINDLE_LASER_ENA_PIN             13  // Pullup or pulldown! RK -> 10 -> D13
    #define SPINDLE_LASER_PWM_PIN             15  // Hardware PWM RK -> 4 -> D15

BUT 13 is also used as #define HEATER_0_PIN                          13  // (extruder)

Since you also have 
#define EXTRUDERS 1
#define TEMP_SENSOR_0 1

This pin will be kept low by the hotend code

rowokii, 2023-09-09:
Yup, that was it!  

After shuffling the pins so the extruder pin is not shared with the ena pin, everything worked. 

Thanks!!!!     Boneheaded user setup error! :p

closed 2023-09-09 by rowokii