MarlinFirmware/Marlin #26258: [Bug] I2C experimental input shaping causes flicker on MKS Tinybee

Reza-v, 2023-09-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The (relatively) new input shaping that was implemented into marlin causes interference with every other I2C pin on the board which can be dangerous considering some of the heating outputs on this board are also on I2C pins.
When enabled, pins do not go high and low in an instant. Instead they go from high, start to flicker and the flicker gets slower and slower until the pin is low. Every time you make a click on the display there is a terrible noise. Input shaping also causes the homing to hang (there is already a thread open about that issue).

Marlin past version 2.0.9 causes layershifts on this board. People have figured out how to fix it, i haven't yet. 

The linear advance on both of the mentioned versions does not work. There is a K value in the firmware and you can change it but it has no effect.

This board is just headaches all over. It isn't very well supported by marlin or any other firmware which is understandable as It isn't very popular but frustrating still.

Another issue is that i can't seem to build the firmware anymore. I definitely remember doing it and uploading it successfully but now i cannot. Even at the time it was inconsistent. It would work when it wanted to, could be because of vsc. Nonetheless, i've attached the logs.
Any help regarding any of the above issues is greatly appreciated.

[vsc log reza-v_045614.txt](https://github.com/MarlinFirmware/Marlin/files/12568280/vsc.log.reza-v_045614.txt)
[config tinybee 2.1.2.1 reza-v_045149.zip](https://github.com/MarlinFirmware/Marlin/files/12568283/config.tinybee.2.1.2.1.reza-v_045149.zip)

### Steps to Reproduce

Enable experimental input shaping and linear advance for i2c boards on the mks tinybee mainboard.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom cartesian

### Electronics

Mks tinybee

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-09-10:
it's I2S, not I2C

thisiskeithb, 2023-09-10:
Duplicate of #25605

thisiskeithb, 2023-09-10:
From https://github.com/MarlinFirmware/Marlin/issues/25605#issuecomment-1495753241:

> IS is not supported with ESP32. My naive guess is the FT is not either. We should add a sanity check.

closed 2023-09-10 by thisiskeithb