MarlinFirmware/Marlin #26259: [BUG] echo:busy:paused for user after M303 ou G29

kramer04, 2023-09-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Tryed Marlin release 2.1.2.1 

SENDING:M303 E-1 C4 S55
//action:notification PID Autotune
PID Autotune start
//action:notification Heating...
 bias: 61 d: 61 min: 54.95 max: 55.57
 bias: 49 d: 49 min: 54.95 max: 55.15
 bias: 49 d: 49 min: 54.94 max: 55.05 Ku: 1075.66 Tu: 10.24
 No overshoot
 Kp: 215.13 Ki: 42.02 Kd: 734.32
 bias: 49 d: 49 min: 54.94 max: 55.07 Ku: 920.65 Tu: 10.40
 No overshoot
 Kp: 184.13 Ki: 35.41 Kd: 638.32
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
//action:notification PID tuning done
#define DEFAULT_bedKp 184.13
#define DEFAULT_bedKi 35.41
#define DEFAULT_bedKd 638.32
//action:prompt_end
//action:prompt_begin M1 Stop
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for u
[config.zip](https://github.com/MarlinFirmware/Marlin/files/12569726/config.zip)
ser
echo:busy: paused for user

### Bug Timeline

with 2.1.2.1

### Expected behavior

no echo:busy:paused for user

### Actual behavior

for example launch M303 for bed , when it ended sent message to pronterface
echo:busy:paused for user

### Steps to Reproduce

M303 E-1 C4 S60

### Version of Marlin Firmware

release stable 2.1.2.1

### Printer model

CR-10 SMART PRO

### Electronics

Stock electronics

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-09-11:
It also shows //action:prompt_begin M1 Stop

Ie Unconditional stop, which would cause "paused for user", so that bit is actually working as it should.

the question is where is the M1 coming from.

kramer04, 2023-09-11:
i tried with no screen, no filament runout, no CR-TOUCH, no power loss recovery...
>>> M303 E-1 C4 S55
SENDING:M303 E-1 C4 S55
PID Autotune start
 bias: 60 d: 60 min: 54.95 max: 55.57
 bias: 50 d: 50 min: 54.95 max: 55.09 Ku: 903.89 Tu: 10.24
 No overshoot
 Kp: 180.78 Ki: 35.31 Kd: 617.06
 bias: 47 d: 47 min: 54.97 max: 55.09 Ku: 947.16 Tu: 11.04
 No overshoot
 Kp: 189.43 Ki: 34.32 Kd: 697.11
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define DEFAULT_bedKp 189.43
#define DEFAULT_bedKi 34.32
#define DEFAULT_bedKd 697.11
echo:Unknown command: "M1M105"

it ended M1M105 so weird.

closed 2023-12-26 by github-actions[bot]