MarlinFirmware/Marlin #26260: [BUG] (bug summary) BTT Mini 12864 V1 has the preset colors in the Light menu mixed.

Lordcheche, 2023-09-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When selecting a preset color from the Lights menu the displayed colors are wrong. The config file is NEO_RGB, but for it to work properly it should be NEO_BRG. When selecting a custom color also from the menu it is bugged too. I know its the menu's fault because the when turning on my printer the default color is displayed correctly and all M150 commands work properly.

### Bug Timeline

Since I started using Marlin

### Expected behavior

When the color red is selected the screen turns red, with blue it becomes blue and the same with green.

### Actual behavior

I select the red preset and the screen displays is blue

### Steps to Reproduce

1. Turn on the printer.
2. Go to change the color thru the display of the printer.
3. It doesn't work. 

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom printer.

### Electronics

MOBO: BTT Octopus 1.1, Display: BTT Mini 12864 V1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/12570921/Config.zip)

thisiskeithb, 2023-09-10:
This is an issue with some STM32H7 boards only and applies to all NeoPixels, not just the ones for this display. You can bypass the sanity check if your board is affected until a fix can be found.

Lordcheche, 2023-09-10:
> This is an issue with some STM32H7 boards only and applies to all NeoPixels, not just the ones for this display. You can bypass the sanity check if your board is affected until a fix can be found.

For my second strip every thing works fine. The problem is that even if i use another RGB config for example BRG (It fixes the presets), then the m150 commands won't work properly

thisiskeithb, 2023-09-11:
The main issue is that this doesn’t affect all STM32H7 boards / configs the same way and there’s not an obvious reason why (like incorrect pins).

Lordcheche, 2023-09-11:
> The main issue is that this doesn’t affect all STM32H7 boards / configs the same way and there’s not an obvious reason why (like incorrect pins).

STM32F4 too? because i dont have a STM32H7 chip

ellensp, 2025-08-15:
Neo pixels require precise timing.  if this is out of spec the data from one colour ends up on the next colour.

For data gathering purposes, what chip is on your BTT SKR 3 EZ,  STM32H743VI or STM32H723VG ?

thurask, 2025-08-15:
STM32H723VG.

thinkyhead, 2026-10-08:
Thanks for the detailed report and config. In current bugfix-2.1.x, the menu presets, custom colors and M150 all send colors through exactly the same code, so the wrong colors look like a signal-timing problem between STM32 boards and these panels' LEDs, not a menu bug. (With NEOPIXEL_M150_DEFAULT -1, M150 also drives your second GRB strip, which may be why it looked correct.) We can't fix this safely for 2.1.3 without a logic-analyzer capture. If you or anyone with an affected board can capture the NeoPixel data line, please attach it here.