MarlinFirmware/Marlin #26270: [FR] make "z probe type" configurable from the menu

vlsi, 2023-09-11:
### Is your feature request related to a problem? Please describe.

Currently, z-probe type has to be selected on the menu, however, sometimes it might be helpful to build a single firmware for several machines that differ slightly.

### Describe the feature you want

I suggest users should be able to activate several supported probe types, and then there should be a menu item for selecting the active probe type.

The same could probably be available for `Use Z probe for homing`. The relevant menu item could appear/disappear depending on the "activated probe type".

EvilGremlin, 2023-09-11:
Such things are directly against Marlin's concept. Possible, but off the table for any forseeable future. I imagine it won't happen at least as long as AVR is supported. Also it's additionsl support and maintenance overhead and we have enough problems with leveling as it is.

vlsi, 2023-09-11:
Frankly speaking, I think testing would be easier if modes could be switched in runtime.

For instance, I can connect both z endstops and a z-probe at the same time on my machine. I can even connect several different probes at the same time (inductive and bltouch). Then switching beetween them would be easier than reflashing the machine for every probe configuration.

I do not suggest that everybody should integrate all the possible probe types into their firmware.

At the same time, the corresponding menu item could appear only in case there are several probe type selected, so it sounds more like pay-as-you-go feature which is in line with Marlin concept.

Of course, the corresponding "setting" should appear o dissappear depending on the "multiple probes selected".