MarlinFirmware/Marlin #26272: Ability to have additional Z-homing probes (for less accurate probes) [FR] (feature summary)

moswald14, 2023-09-11:
### Is your feature request related to a problem? Please describe.

I have a knock off BLTouch and I'm always having to readjust my Z-offset for each print. I believe this is because It gets an inaccurate home position from the only 2 probes it makes when it homes.

### Describe the feature you want

Add an int to the code that dictates the number of homing probes and extra probes it does and take the average, for a more accurate home.

classicrocker883, 2023-09-25:
this may be an issue with the probe itself.

but if you wanted to increase the number of probes, go to Configuration.h and look for `MULTIPLE_PROBING`

Uncomment that and change it to a number more than 2, also you can try defining `EXTRA_PROBING`

you can try doing M48 and see what the result is. do a test using 3 or 4 probes, then one with 7 or 10. 
you should not have a number greater than 0.01 deviation.

I would try adding some kind of plastic safe liquid lubricant to the probe, perhaps something is getting stuck giving wrong results. or replace the plastic tip.

ellensp, 2023-09-25:
Already implemented

closed 2023-09-25 by ellensp