MarlinFirmware/Marlin #26273: [BUG] (powerloss not working and breaking run of print)

bart-tech3d, 2023-09-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

if I turn on the powerloss recovery function in the menu, the printer starts to jam during printing, sometimes the display goes black. However, after a power outage, the printer will not offer the option to continue printing.

### Bug Timeline

After compile Marlin 2.1.x bugfix

### Expected behavior

After the power supply is interrupted, a "PLR" file is saved on the SD card or USB flash drive, which the printer records at start-up and prompts to continue printing.

### Actual behavior

The "PLR" file probably won't even be created, the printer will start jamming, and anyway, the powelsoss function doesn't seem to be there. I also tried compile only with SD, but this did not help.
With Marlin 2.0. no problem with Robin nano V2.0.

### Version of Marlin Firmware

Marlin-2.1.x bugfix

### Printer model

own model

### Electronics

MKS robin nano V3.0 or 3.1 + TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I can't give a LOG output because I can't connect to the serial port to pronterface (printrun) with the nano V3.0 board.
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/12580728/config_files.zip)

firepower9966, 2023-12-10:
Are you using MKS PWC.
https://a.aliexpress.com/_mMDuwNS

shabgarde, 2024-01-21:
@thisiskeithb 
I have the same issue. Please share if you have any solution

bart-tech3d, 2024-02-01:
not yet, but I would like to try pwc. But first, I will try to simulate it by manually connecting the powerloss pin activation during printing and then turning off the printer.

bart-tech3d, 2024-07-10:
I integrated to PSU MKS PWC V3.1. The shutdown works, the PLR ​​file is saved in the event of an outage.
But there were two problems:
1. absolute ignoring of PLR file (both SD and USB disk). He acts like he doesn't exist
-I solved this by editing cardreader.cpp , where I commented out the original // Check for PLR file.
 //TERN_(POWER_LOSS_RECOVERY, if (recovery.check()) do_auto = false); and put the following lines in the " if (stat) { " condition:
// Check for PLR file after mounting the media
 DEBUG_ECHOLNPGM("Checking for PLR file...");
 recovery.check(); // Check for PLR file, this will trigger the existing behavior if found 

Only then did the printer respond to the presence of the PLR ​​file

2. The restore starts from the beginning as if the last position was not saved and I can't figure out why.

bart-tech3d, 2024-07-10:
my files

[power_loss_my_files.zip](https://github.com/user-attachments/files/16167699/power_loss_my_files.zip)

closed 2024-10-23 by github-actions[bot]