MarlinFirmware/Marlin #26274: [BUG] SKR mini E3 DIP V1.0 - printing is stuttering on same position

MeloenBe, 2023-09-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After update 2.0.9.3 , whenever I try to print something with fast movements, the printer is stuttering massively.
This always happens at the same spot in the print (between left down corner and left middle spot). Also the LCD screen becomes very slow.
I already tried every new update after it but it keeps happening. Also used different mirco SD's, reposition the 3D print and updated slicer but nothing helps.
The higher the Z-value goes, the more less the stuttering becomes. After 10mm, the printer prints normal.
[confi.zip](https://github.com/MarlinFirmware/Marlin/files/12584341/confi.zip)

It seems some issue with bedleveling but I'm not sure and didn't test it yet
I downgrade to version 2.0.9.3 and the problem is gone.

### Bug Timeline

Started from 2.0.9.4

### Expected behavior

Print normally

### Actual behavior

Prints normally and then suddenly begins to stutter until it passes some points and then prints normal again. It stutters in the same area until Z is more than 10mm.

### Version of Marlin Firmware

2.0.9.4

### Printer model

Creality Ender 3 Pro

### Electronics

BigTreeTech SKR mini e3 DIP V1.0 with TMC2209

### Add-ons

BLtouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[confi.zip](https://github.com/MarlinFirmware/Marlin/files/12584351/confi.zip)
https://github.com/MarlinFirmware/Marlin/assets/144733530/4469e066-2240-43cc-b57a-c44671a210ef

EvilGremlin, 2023-09-12:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

Did you really?
This is problem with leveling because 10mm is deafult FADE_HEIGHT, so try UBL too. Maybe bilinear didn't like junction deviation at that point. Also check slicing resolution and gcode - i suspect a lot of stupidly short segments, like 0.001mm

MeloenBe, 2023-09-12:
Yes I tried bugfix-2.1.x

Also why doesn't this happen with version 2.0.9.3 and it does with higher versions?
Version 2.0.9.3 prints perfectly so slicing and gcode should be fine.

I'll try UBL tonight

EvilGremlin, 2023-09-12:
probably becasue JUNCTION_DEVIATION is default since 2.0.9.4

MeloenBe, 2023-09-12:
I also have JUNCTION_DEVIATION activated on 2.0.9.3

Here are the files from that version

[confi 2.0.9.3.zip](https://github.com/MarlinFirmware/Marlin/files/12584723/confi.2.0.9.3.zip)

MeloenBe, 2023-09-13:
Ok, it seems that ABL calculates or does something different after version 2.0.9.3

I did the following tests yesterday:

Version bugfix-2.1.x with ABL Bilinear (25 points) => failed
Version 2.0.9.3 with ABL Bilinear (25 points) => succeed
Version bugfix-2.1.x with ABL Bilinear (49 points) => failed
Version 2.0.9.3 with ABL Bilinear (25 points) => succeed
Version bugfix-2.1.x with UBL (100 points) => succeed
Version 2.0.9.3 with ABL Bilinear (25 points) => succeed
Version bugfix-2.1.x with UBL (100 points) => succeed

I'll be using UBL then. Thanks for the input!

MeloenBe, 2023-09-20:
Found the problem... after getting the same error again, even with UBL
By default I always increased the buffer size:
#define BLOCK_BUFFER_SIZE 64

Somewhere in another thread, about SKR boards, someone said to increase the slowndown, so I did:
#define SLOWDOWN_DIVISOR 16

After this my problem stopped.
JD works fine again, on UBL and ABL.

**Maybe it is an idea to write a comment at "BLOCK_BUFFER_SIZE" to increase the slowdown if the buffer gets increased.**
This is at the moment only commented at the slowdown

radry, 2024-01-13:
I have the same problem (although on a SKR Mini E3 **V1.2** ). On Marlin 2.0.9.1 there was no issue but after updating to 2.1.2.1 every print stutters at speeds above 50mm/s at every direction change. LCD also becomes very slow. The Config was manually matched to old one, so no new features should be enabled which could cause this.  
Junction deviation is not the cause, it was also enabled in my 2.0.9.1 build. ABL and UBL not enabled/used.
   
Reading the previous comment, I tried the following in the config:
BLOCK_BUFFER_SIZE 64
SLOWDOWN_DIVISOR 16
  
**Which did NOT solve the issue.**

Anyway this is still a big issue, since 2.0.9.1 worked fine and suddenly the printer is useless.

MeloenBe, 2024-01-18:
Nice to see that someone else has a similar problem. I kept this issue open because it also didn't get fixed with me.
 The settings
BLOCK_BUFFER_SIZE 64
SLOWDOWN_DIVISOR 16
did do a bit about the printing issue but it never went away...
The problem did only occur when I was printing 1) a very detailed piece with many rapid movements or 2) a big file +- 200mb

I tried everything (bug fix with ABL and UBL) but it kept happening. I have 2x Ender 3 Pro now.
1 stock with 4.2.2 board
1 with BigTreeTech SKR mini e3 DIP V1.0 with TMC2209

And I can say that the stock has only 10-15% of doing this stuttering (can't confirm the slow LCD response) compared to the other one. The stuttering also happens with any version after 2.0.9.3 on the stock one. The version 2.0.9.3 is the last stable one. No stuttering at all, fast printing and no LCD problem.

@radry , please try marlin 2.0.9.3 because this seems to me the last update that everything goes well. Maybe try the 2.0.9.4 after that and let me know the result.

dh28567, 2024-01-22:
I was having the same exact issue in 2.1.2.1 and switched to 2.0.9.1 and it fixed it

radry, 2024-01-22:
@MeloenBe 
I give up. I tried 2.0.9.4 and even went back to 2.0.9.1 but I suddenly had the same issue in that version although never before. So I started to suspect the slicer because that's also an element that I updated at the same time. So I went back to Cura 4.12 (last known working version for me) from 5.6 but the issue persisted (maybe the new versions settings "poisoned" the old version? I have no idea if and how cura shares settings between versions). I even tried going back to classic jerk, just in case. I also reseted the eeprom every time, to be sure.  I'm at a loss, no idea what happened. I ruled everything out that could be a potentional cause.  

The choppy movements starts at the second fast layer, so the first ones goes fine then it seems the "command queue" is full and the board can't keep up processing or delivering the commands. What is strange is that changing the settings as mentioned in my previous comment didn't have any effect at all. Maybe as a last ditch effort I'll try a new SD card (not for the firmware but the print files), could be that it got corrupted and thus can't be read fast enough?

MeloenBe, 2024-01-25:
The bug started in 2.0.9.4 , so someone should check deeply in the code what changed with 2.0.9.3

It has to do something with the leveling part because after the fade height (10mm for me) is reached, the print goes fine.
The closer I get to the 10mm , the less the stuttering becomes.
As mentioned, it seems only to happen in **the outer + inner layer** but certainly not in the bottom/fill layers! These go fine.

@radry
Start from a clean version of 2.0.9.3 and format your SD card (slow format!)
I don't think the slicer is the problem. In my case I kept using the same sliced model on each version to test in which version the bug started.
Do you use ABL or UBL?

closed 2024-07-01 by thinkyhead