MarlinFirmware/Marlin #26276: [BUG] Random "Endstops" message

classicrocker883, 2023-09-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Not sure exactly how it came about, I'll try to reproduce it.

So out of nowhere I get a status message saying "Endstops &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; chrP" - (it had many spaces inbetween)

I traced it to **module/endstops.cpp** line ~351
```y
void Endstops::event_handler() { ...

    TERN_(HAS_STATUS_MESSAGE,
      ui.status_printf(0,
        F(S_FMT GANG_N_1(NUM_AXES, " %c") " %c"),
        GET_TEXT(MSG_LCD_ENDSTOPS),
        NUM_AXIS_LIST_(chrX, chrY, chrZ, chrI, chrJ, chrK, chrU, chrV, chrW) chrP
      )
    );
```

[image]

as you can see there is some sort of error popping up.

### Bug Timeline

possibly old

### Expected behavior

I'm guessing this should not have popped up??

### Actual behavior

Popped up randomly

### Steps to Reproduce

1. Using ProUI
2. Have everything unplugged from the mainboard except power and LCD cable
note: this is probably just a fluke due to everything unplugged. ie: no endstops were being registered.

but seeing the "<error-type>" in VSCode when hovering over the function in question made me submit this.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### Electronics

G32 mainboard

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-09-13:
Please attach your configuration files.

classicrocker883, 2023-09-16:
I believe I have traced this to **lcd/marlinui.h** line ~514

the newer version:
```y
  template<typename... Args>
  static void status_printf(int8_t level, FSTR_P const ffmt, Args... more) { status_printf_P(level, FTOP(ffmt), more...); }
```

an older version which didn't show the "error-type"
```y
  static void status_printf(int8_t level, FSTR_P const fmt, ...);
```

classicrocker883, 2023-12-04:
I wonder would changing....
```diff
-     ui.status_printf(0,
-       F(S_FMT GANG_N_1(NUM_AXES, " %c") " %c"),
        GET_TEXT(MSG_LCD_ENDSTOPS),
        NUM_AXIS_LIST_(chrX, chrY, chrZ, chrI, chrJ, chrK, chrU, chrV, chrW) chrP
      );

+      FTOP(S_FMT GANG_N_1(NUM_AXES, " %c") " %c"),
+      FPSTR(S_FMT GANG_N_1(NUM_AXES, " %c") " %c"),

+     ui.status_printf_P(0,
```
the above code isn't verbatim, just as an example to replace `F` with `FTOP` or `FPSTR`,  if with `FTOP`, then this must be `ui.status_printf_P`

classicrocker883, 2023-12-05:
or what about something like this?
```y
      ui.set_status_and_level(MString<30>(F(S_FMT GANG_N_1(NUM_AXES, " %c") " %c"), GET_TEXT_F(MSG_LCD_ENDSTOPS),NUM_AXIS_LIST_(chrX, chrY, chrZ, chrI, chrJ, chrK, chrU, chrV, chrW) chrP));
```

closed 2024-06-15 by github-actions[bot]