MarlinFirmware/Marlin #26277: [BUG] (Biqu BX not properly working)

digant73, 2023-09-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I recently updated the firmware for Biqu BX to latest available version but it seems something is no more properly working.:
- At boot, My external BTT TFT35 is no more able to connect to the mainboard. In order to connect the TFT, from the TFT menus I have to press on the disconnect button and then to exit from that (the TFT is finally able to connect to mainboard).
- The rotary encoder on the BX's 7" TFT display is not properly working. It jumps back and forth skipping items in the menu

Attached my configuration. They have been working in the previous fw versions.

### Version of Marlin Firmware

2.1.2.1

### Printer model

biqu BX

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/12596297/Marlin.zip)

thisiskeithb, 2023-09-13:
> The rotary encoder on the BX's 7" TFT display is not properly working. It jumps back and forth skipping items in the menu

Uncomment `ENCODER_PULSES_PER_STEP` and set it to `5`.

> At boot, My external BTT TFT35 is no more able to connect to the mainboard. In order to connect, from the TFT menus I have to press on the disconnect button and then to exit from that (the TFT is finally able to connect to mainboard).

So you have two TFTs connected? Stock (that is controlled by Marlin) and an external one running BigTreeTech's TFT firmware?

Reduce the baudrate for serial port 1 to 115200.

Both of these values come from the Biqu BX config we maintain: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/BIQU/BX

I use this config on my Biqu BX without issue.

digant73, 2023-09-13:
Ok thanks Keith. I will try the changes although my settings were working in the past.

Yes I have also an external TFT35 just for testing the BTT TFT fw.
So
`#define BIQU_BX_TFT70`
is no more enough to also set the proper settings for the rotary encoder for the BX's TFT?

thisiskeithb, 2023-09-13:
> So
#define BIQU_BX_TFT70
is no more enough to also set the proper settings for the rotary encoder for the BX's TFT?

The `BIQU_BX_TFT70` define tells Marlin that you have this TFT connected and defines the associated pins.

You’ll override `ENCODER_PULSES_PER_STEP` since the default of `4` can be too coarse like you’ve found.

digant73, 2023-09-13:
ok, the suggested changes worked. Thanks!
Closing the ticket

closed 2023-09-13 by digant73