MarlinFirmware/Marlin #26279: [BUG] Non-constant feedrate for G2/G3

rnj1, 2023-09-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

This is related to https://github.com/MarlinFirmware/Marlin/issues/15295, which is still not fixed. 
I tested it on Marlin 2.1.2.1 and the current 2.1.x bugfix (14.9.2023 - config. h ver 02010300). 
When using G2/G3, the feedrate is overshooting and not constant. For testing I tried a full arc with verious speeds and radius (0.5-10mm) e.g. "G2 I2 J0" and speeds (60-3000mm/min), as I tried many different configurations (MIN_CIRCLE_SEGMENTS(0-0.1), MAX_ARC_SEGMENT_MM (0.1- 1), N_ARC_CORRECTION (25-72), MIN_CIRCLE_SEGMENTS(8-72+ commented)) always with the same behaviour. 
The issue with the G2/G3 arc feedrate implementation is still not fixed. With all settings I tried the feedrate overshoots leading to artefacts when printing. The feedrate is only computed correctly when only one axis moves.
I analyzed the step/dir outputs of my DUE board with a scope and Matlab and it shows an cyclic half-wave overshoot of ~35%: 
[image]
The acutal trajectory seems to be fine:
[image]

### Bug Timeline

At least Marlin 2.0 in 2019?

### Expected behavior

The toolspeed during G2/G3 arc motion should stay constant

### Actual behavior

During arc motion, the toolspeed is overshooting 35% in an arc like behavior:
[image]

### Steps to Reproduce

[CfgBgfx.zip](https://github.com/MarlinFirmware/Marlin/files/12607387/CfgBgfx.zip)

### Version of Marlin Firmware

2.1.2.1 and 'FIRMWARE_NAME:Marlin bugfix-2.1.x (Sep 14 2023 10:48:50) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff↵'

### Printer model

Custom

### Electronics

DUE with RAMPS-FD-V2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

DerAndere1, 2023-09-15:
- Is the speed accuracy better when moving at lower feedrate or when you increase DEFAULT_ACCELERATION  DEFAULT_TRAVEL_ACCELERATION  and DEFAULT_MAX_ACCELERATION for the axes involved?

rnj1, 2023-09-15:
I tried speeds from 1mm/s to 50mm/s and with higher speeds the relative speed error stay the same (2-3%). I also reduced the accerlation from 1000mm/s^2 to 500mm/s^2 which looked the same (didn't make a scope measurment though and also I didn't try less yet) and also tested S-curve accelration which had, for printing, the same appearance.
Here the same Trajectory with G2 at 5mm/s and 25mm/s. It's 2% (+/-0.1mm/s) @5mm/s and 3.2% (+/- 0.8mm/s) @ 25mm/s.
[image]
[image]

ellensp, 2023-09-15:
Have yo actually try bugfix?
Many bug fixes where not put into 2.1.2.1, only a select few

rnj1, 2023-09-15:
I tried also yesterday's bugfix (Marlin bugfix-2.1.x (Sep 14 2023 10:48:50)), which does have the same issue (In the first few charts with "BGFX" in the title, it's a measurement with that bugfix version.

stef-evotech, 2023-10-12:
Hi!

I observed some similar issues, could you report your Jerk/Acceleration values? Such issues can also be due to older Stepper-Drivers like A4988 because they sometimes have a torque-irregularity. Also I noticed some weird bumps when using newer Trinamic-drivers with high microstepping and microstep-interpolation on!

Also what's quite simliar looking are backlash-problems, but i figure you checked your pulleys/belts etc,...

Cheers

rnj1, 2023-11-10:
The charts were extracted from the scope measurement of the output of the 32bit microcontroller without any driver connected. I used the standard settings for jerk as written in my config. For accelereation I tried various values. Also my printer has 3x precision linear drives with optical encoders (JennyScience Lxs / XVi75V8) without any backlash...
To me it's a numeric bug of Marlin

stef-evotech, 2023-11-13:
I had a quick look at your config. Acceleration/Jerk seems reasonable, what seems off - at least in the version you uploaded - is that CLASSIC_JERK is disabled so in this configuration you may be actually using JUNCTION_DEVIATION. Please check if that is still the case.

XDA-Bam, 2023-12-19:
> I had a quick look at your config. Acceleration/Jerk seems reasonable, what seems off - at least in the version you uploaded - is that CLASSIC_JERK is disabled so in this configuration you may be actually using JUNCTION_DEVIATION. Please check if that is still the case.

Agree. This may be an artifact of the default JD implementation. @rnj1 Please be so kind and re-test using `CLASSIC_JERK`. If that solves it, it would be nice to see if running JD again but with the additional config entry `#define JD_USE_LOOKUP_TABLE` also gets rid of the fluctuation. If `CLASSIC_JERK` doesn't solve it, it would be time to investigate the code for G2/G3.