MarlinFirmware/Marlin #26281: [BUG] screen freeze on entering settings during print

vovodroid, 2023-09-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have Biqu B1 SE with SKR 2 + TFT SPI screen. On latest Marlin [[cron] Bump distribution date (2023-08-26)](https://github.com/MarlinFirmware/Marlin/commit/11f98adcce51f06a85ef0dd64e6e6b16139a34bd) with both https://github.com/MarlinFirmware/Marlin/commit/fef772ef29e502c8fd2f53e10de383dbcb3529fe and https://github.com/MarlinFirmware/Marlin/commit/3ea87ad54c1e445a1c8cfef1489809edae5decb3 reverted when I try to enter Settings menu by clicking on touch screen or rotary knob sometimes screen is getting black or freeze for about 10 secs. And sometimes it freezes on navigating inside Settings menu (for instance while trying to reach K tune).

### Bug Timeline

Somewhere after https://github.com/MarlinFirmware/Configurations/commit/7fa8b7f1244dbadd3d702e9c637529ca845c3139

### Expected behavior

Setting menu works

### Actual behavior

Screen is getting black or freezes

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/commit/11f98adcce51f06a85ef0dd64e6e6b16139a34bd

### Printer model

Biqu B1 SE

### Electronics

Stock SKR 2 + TFT SPI

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Config:
[B1 SE.zip](https://github.com/MarlinFirmware/Marlin/files/12608362/B1.SE.zip)

Gcode file screen freezes while printing (first print after updating Marlin, I didn't print other files):

[0914-1318-Tsirkuly~BaseBody-1h35m.zip](https://github.com/MarlinFirmware/Marlin/files/12608372/0914-1318-Tsirkuly.BaseBody-1h35m.zip)

classicrocker883, 2023-09-14:
I had an issue with `#define ENCODER_WAIT_MS 20`  
this is in **lcd/e3v2/common/encoder.h**

but having it changed to `10` ms definitely helped. 
this is for a DWIN LCD like on the Ender3V2.

im not sure how it can be related to your screen, that is Idk if it uses the same encoder files, but I would start there.

vovodroid, 2023-09-15:
> im not sure how it can be related to your screen,

I think it's not related. I have TFT SPI screen, and encoder.h even isn't compiled.

vovodroid, 2023-09-15:
I don't know whether it's related, but when no printing clicking on return in settings screen throws to move Z axis screen:

[image]

[image]

Could this issue be linked to  **Color UI remaining time** #24843 or some remains of  **Add support for SPI TFT displays and touch screen for STM32H7 MCUs** #25784 (though I reverted two related commits)?

closed 2023-11-17 by vovodroid