MarlinFirmware/Marlin #26287: Sainsmart 2004 I2C LCD line break

borisz94, 2023-09-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sainsmart 2004 I2C LCD line break not correct. Long lines overflow and move to other places on the display (images). This happens in all languages.

### Version of Marlin Firmware

Marlin 2.1.x bugfix

### Electronics

Megatronics V2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Images.zip](https://github.com/MarlinFirmware/Marlin/files/12641317/Images.zip)
[configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/12641318/configuration.files.zip)

ellensp, 2023-09-17:
Himm...  this seems rather similar to https://github.com/MarlinFirmware/Marlin/issues/26148 
 which was meant to be fixed with https://github.com/MarlinFirmware/Marlin/pull/26264

ellensp, 2023-09-17:
These error do not seem to be happening in current bugfix
Update your bugfix code and try again.

closed 2023-09-17 by ellensp

thisiskeithb, 2023-09-17:
> which was meant to be fixed with #26264

You had it right the first time 😄 This was fixed in https://github.com/MarlinFirmware/Marlin/pull/26154.

https://github.com/MarlinFirmware/Marlin/pull/26264 is a PR for `.editorconfig` changes.

ellensp, 2023-09-17:
fixed failed cut and pate.

borisz94, 2023-09-18:
Thanks.  The new version really solved the problem.  Sorry for the new bug report, but I couldn't find the solution in the search engine.