MarlinFirmware/Marlin #26289: [BUG] No SD automount

vovodroid, 2023-09-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On BTT SKR 2 board with both SD and USB SD card is not mount on insert. Mounting via menu is needed.

Probably it causes  or has the same root as **[BUG] No resume after power loss #26280**.

### Steps to Reproduce

Boot printer.
Insert SD card. Nothing happens.
Remove it, insert USB disk. It's mounted.
Remove USD disk, insert SD.
Now SD is mounted and you can print from SD, whether by clicking on SD card icon or via menu.

### Version of Marlin Firmware

Marlin bugfix 2.1 2023-08-26

### Printer model

Biqu B1 SE

### Electronics

BTT SRK 2 board

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[B1 SE.zip](https://github.com/MarlinFirmware/Marlin/files/12642612/B1.SE.zip)

vovodroid, 2023-12-05:
Now I have SKR-3 board, without USB flash disk, but still there is no SD automount on insert. Latest (2023-12-02) Marlin, attached config files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13559556/Marlin.zip)

thisiskeithb, 2023-12-05:
The SKR 2 & 3 don't have an `SD_DETECT_PIN` for the onboard SD, so no auto-mounting can occur.

closed 2023-12-05 by thisiskeithb

vovodroid, 2023-12-05:
Isn't it worth to add pooling? Anyway CPU is not busy while not printing.

vovodroid, 2023-12-05:
Or at least auto mount/refresh upon clicking card icon on touch screen or "Print from media" menu? It'll provides the same effect as auto mount.