MarlinFirmware/Marlin #26290: [BUG] Ender 3 Pro fails to build

Lazer-Khat, 2023-09-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

PS C:\Users\Willi\Downloads\Marlin-bugfix-2.1.x> platformio run --silent -e STM32F103RE_creality
Marlin\src\MarlinCore.cpp: In function 'void manage_inactivity(bool)':
Marlin\src\MarlinCore.cpp:677:20: warning: unused variable 'e_stepper' [-Wunused-variable]
  677 |       const int8_t e_stepper = TERN(HAS_SWITCHING_EXTRUDER, active_extruder >> 1, active_extruder);
      |                    ^~~~~~~~~
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\temperature.cpp:30:
Marlin\src\module\temperature.cpp:301:14: error: static assertion failed: HEATER_0_MAXTEMP (315) is too high for thermistor_1.h with HOTEND_OVERSHOOT=20.
  301 |     S >= 998 || M <= _MAX(TT_NAME(S)[0].celsius, TT_NAME(S)[COUNT(TT_NAME(S)) - 1].celsius) - HOTEND_OVERSHOOT, \
Marlin\src\module\../inc/../core/macros.h:632:24: note: in definition of macro 'EVAL1'
  632 | #define EVAL1(V...)    V
      |                        ^
Marlin\src\module\../inc/../core/macros.h:631:30: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:630:24: note: in expansion of macro 'EVAL2'
  630 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:24: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:30: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:630:30: note: in expansion of macro 'EVAL2'
  630 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:629:24: note: in expansion of macro 'EVAL4'
  629 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:24: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:30: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:630:24: note: in expansion of macro 'EVAL2'
  630 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:24: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:30: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:630:30: note: in expansion of macro 'EVAL2'
  630 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:629:30: note: in expansion of macro 'EVAL4'
  629 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:628:24: note: in expansion of macro 'EVAL8'
  628 | #define EVAL16(V...)   EVAL8(EVAL8(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:24: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:631:30: note: in expansion of macro 'EVAL1'
  631 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:630:24: note: in expansion of macro 'EVAL2'
  630 | #define EVAL4(V...)    EVAL2(EVAL2(V))
[... 91 lines cut ...]
Ender 3 Pro

### Electronics

4.2.7 board, Sprite Extruder Pro w/ CR Touch

### Add-ons

Dual Z Belt

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Lazer-Khat, 2023-09-17:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12643835/Configuration.zip)

thisiskeithb, 2023-09-17:
> error: static assertion failed: HEATER_0_MAXTEMP (315) is too high for thermistor_1.h with HOTEND_OVERSHOOT=20.

Reduce your `HEATER_0_MAXTEMP` since it's higher than the max temperature for your thermistor.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-09-17 by thisiskeithb

Lazer-Khat, 2023-09-17:
That seemed to work! New question: which environment should I use for a 4.2.7 board?
[image]