MarlinFirmware/Marlin #26291: [BUG] Screen stays blue after firmware update Main and Bugfix 2.1

M-Wahren, 2023-09-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
Downloaded latest main version and configured setting and tried upgrading my printer (Ender-3)
I ran autobuild marling in VS Code using "STM32F103RE_btt" as i have upgraded my motherboard to BTT mini E3 v2.0.
After flashing my board using SD card i get blue screen on my display and nothing more after 30-40 seconds..

I then tried running the same settings but using latest bugfix but same thing..
Tried connecting using my octoprint and checking to see if i can controll it without my screen but no success, cant even gett connection..

I have also tried compied using btt_USB with no success, still blue screen..

I have now tried the outdated btt_maple build and now i gett a screen. Unfortunately when pressing my wheel i seem to gett an error and my printer freezes up and reboots.. (same on main and bugfix)
Anyone seen this before?

Thanks in advance!

### Bug Timeline

Saturday when i downloaded a new firmware

### Expected behavior

Boot

### Actual behavior

Nothing stuck on firmware install

### Version of Marlin Firmware

02010201 and 02010300

### Printer model

Ender 3

### Electronics

BTT mini E3 v2.0

### Add-ons

Bltouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/12651548/Marlin-2.1.x.zip)

HeartOfGermany, 2023-09-19:
Are you sure "SERIAL_PORT 2" is correct? I have mine defined as 1 and if I recall correctly with 2 it did not work. I could be wrong, but I thought 2 is for motor controller communication. Just a guess on my side. Before trying, I recommend checking against other ready to use configs, which you should find easily in the wilds.

ellensp, 2023-09-19:
On the BTT SKR MINI E3 V2.0 there are 3 serial ports
-1 is he usb serial cable
1 is the serial port on the exp 1 header (you do not want to enable this when you have a CR10_STOCKDISPLAY which is also plugged into exp1)
2 is the serial port on the TFT header

ellensp, 2023-09-19:
Tested on Current bugfix 

With these exact configs (from the marlin Configurations)

The screen and board start up as expected.

I suspect the issue is that you selected STM32F103RE_btt, without checking what CPU you have on your controller

If the Controller has a STM32F103 RET6  then you use STM32F103RE_btt
but if it has a STM32F103 RCT6  Then you must use STM32F103RC_btt

closed 2023-09-19 by ellensp

M-Wahren, 2023-09-23:
So it seams that i was using the incorrect CPU but unfortunately it did not solve my problem...

My problem is still there. I went through and tried all 4 variants of the btt auto compiler for RCT6. btt, btt_maple, btt_USB and btt_USB_maple without any success.

i even downloaded the "config.zip" you uploaded with no success..
Tried it with btt and btt_maple and no screen..
so the problem seam to still be there..

ellensp, 2023-09-23:
Since I have tested this hardware combination, twice now, and it still works as expected that leaves hardware..

So perhaps you  have miss identified your motherboard or lcd type?  

regardless, this is not a bug, so this is a support matter

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).