MarlinFirmware/Marlin #26295: [BUG] Error while compiling LPC1768 with tmcsteppers and bltouch

Rastafaraaj, 2023-09-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Cannot compile Marlin 2.1 bugfix after adding TMCSteppers library, without TMCSteppers everything goes fine. 
Error:
```
In file included from Marlin/src/module/../HAL/shared/servo.h:76,
                 from Marlin/src/module/servo.h:29,
                 from Marlin/src/MarlinCore.cpp:140:
Marlin/src/module/../HAL/shared/../LPC1768/Servo.h:51:10: fatal error: Servo.h: No such file or directory

***************************************************************
* Looking for Servo.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Servo.h"
* Web  > https://registry.platformio.org/search?q=header:Servo.h
*
***************************************************************

   51 | #include <Servo.h>
      |          ^~~~~~~~~
compilation terminated.
*** [.pio/build/LPC1768/src/src/MarlinCore.cpp.o] Error 1
```

Tried to enable/disable longpath under Windows 11, tried on Windows 10, tried on newest Ubuntu, still same error exists. 
My configuration is:
LPC1768, TMCSteppers TM2130 UART mode, configuration werk before adding BLTOUCH (only removed // before define BLTOUCH)

### Expected behavior

I expected to compile newest version of Marlin with BLTouch

### Steps to Reproduce

Compile LPC1768 with BTT SKR V1.3 mobo and BLTouch enabled then try to install TMCSteppers library and try again to compile.
[config_drukarka.zip](https://github.com/MarlinFirmware/Marlin/files/12689187/config_drukarka.zip)

### Version of Marlin Firmware

2.1 bugfix

### Printer model

Custom

### Electronics

SKR V1.3, TMC2130

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Rastafaraaj, 2023-09-21:
After 2 days of sitting at front of monitor i found a solution. Maybe its not a bug but i think its a big problem for typical Marlin user.

Resolved by adding "Servo" in platformio.ini
lib_deps = 
	teemuatlut/TMCStepper@^0.7.3
	olikraus/U8g2@^2.35.6
	Servo

and changing 
[common]
       arduino-libraries/Servo@^1.2.1

to 
[common]
       servo

that's it, now its compiled

thisiskeithb, 2023-09-21:
> Resolved by adding "Servo" in platformio.ini
> lib_deps =
> teemuatlut/TMCStepper@^0.7.3
> olikraus/U8g2@^2.35.6
> Servo
>  
> and changing
> [common]
> arduino-libraries/Servo@^1.2.1
> 
> to
> [common]
> servo

Your `platformio.ini` is severely malformed & deviates from what we have in this codebase.

Start with the stock [`platformio.ini`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/platformio.ini) and only change `default_envs` from `mega2560` to `LPC1768`.

```diff
diff --git a/platformio.ini b/platformio.ini
index e3bdb6f..eaf82f8 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = LPC1768
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

This compiles fine.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-09-21 by thisiskeithb