MarlinFirmware/Marlin #26298: [FR] Hide gcode file extension on UI

vovodroid, 2023-09-22:
### Describe the feature you want

Due to limited LCD screen space it may be handy to have option for showing file name only, without ".gcode".

vlsi, 2023-10-02:
What if there are several files with the same name and different extensions?
If Marlin attempts to show at least some of the chars in the extension, then the user could be able to distinguish the files.

What is the purpose of completely hiding the extension?

vovodroid, 2023-10-02:
> What if there are several files with the same name and different extensions?

Is it real case? Both Prusa and Cure produce .gcode files, I believe also other slicers. So it can happen only if use slicers that use different extensions and slice the same STL.

> What is the purpose of completely hiding the extension?

Save the screen space to show as much of name as possible. For example FreeCad saves files with long names like "Unnamed-BodyRevolution.stl".

PanosPetrou, 2023-10-05:
vovodroid, I don't see the point, you could just enable //#define SCROLL_LONG_FILENAMES
in Marlin Configuration. You could even rename the file and remove the extension, before giving the file to the printer. 
Also, as vlsi pointed out, it could cause confusion.

vovodroid, 2023-10-05:
> just enable //#define SCROLL_LONG_FILENAMES

Yes, it's enabled but I don't how it works.

> You could even rename the file and remove the extension, before giving the file to the printer.

Marlin shows only file starting from g: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/sd/cardreader.cpp#L227

thinkyhead, 2023-10-06:
This is an OK idea, and one which I have been contemplating also. When there is a variation in user preference we just make changes like this into an optional setting. Also, since hiding file extensions would increase the code size by a small amount for all builds with an LCD, it would just be nicer to make this change into an optional feature.

vovodroid, 2023-10-07:
Well, actually file name is cut by right screen edge, at least with scrool long names enabled. This is how this option works?

thisiskeithb, 2023-10-07:
> This is how this option works?

If the file name doesn’t fully scroll, it’s probably a bug with that particular UI implementation.

Which LCD/UI are you using?

vovodroid, 2023-10-07:
```
#define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B
#define BTT_TFT35_SPI_V1_0

#define LONG_FILENAME_HOST_SUPPORT
#define SCROLL_LONG_FILENAMES
```

thisiskeithb, 2023-10-07:
> #define BTT_TFT35_SPI_V1_0

I assume `TFT_COLOR_UI` then?

thinkyhead, 2023-10-07:
It looks like that screen doesn't yet implement `SCROLL_LONG_FILENAMES`.

vovodroid, 2023-10-07:
> I assume `TFT_COLOR_UI` then?

Yes:
`#define TFT_COLOR_UI`

vovodroid, 2023-10-07:
So it seems that there is no need to hide extension, it's cut automatically without `SCROLL_LONG_FILENAMES`

thisiskeithb, 2023-11-03:
> It looks like that screen doesn't fully implement `SCROLL_LONG_FILENAMES`.

`SCROLL_LONG_FILENAMES` works perfectly fine with this TFT (more specifically, `TFT_COLOR_UI`).