MarlinFirmware/Marlin #26305: Ender3 V2 Display in Octopus Pro 1.0

gsuresh2u, 2023-09-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, today i tried to install and connect to Ender3 V2 display to Octopus Pro 1.0
Downloaded bugfix branch, and customized as per my printer
done wiring same as per **[](https://github.com/MarlinFirmware/Marlin/pull/24983)**
compiled and installed to board, after installing, screen is just flickkering
i double checked wiring and firmware installed fine 

attaching my config file, can anyone please help ?
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/12708612/Configs.zip)

### Bug Timeline

Today

### Expected behavior

Need to show Display well

### Actual behavior

Screen is Just Flickering

### Steps to Reproduce

1. Custom wiring to ender3 V2 display as per https://github.com/MarlinFirmware/Marlin/pull/24983
2.  Modify and install firmware
3. After firmware installed, Screen is just Flickkering (no display at all)

### Version of Marlin Firmware

Bugfix 2.1.x, downloaded today

### Printer model

Ender 3V2

### Electronics

Octopus Pro 1.0

### Add-ons

Auto Shudown

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-09-24:
### this github is still not a support forum.. 

The tft port is uart 1 on your controller, So you cannot set #define SERIAL_PORT 1  (use a different port) 
You must add #define LCD_SERIAL_PORT 1   or marlin takes a guess at 3 and warns you.
"Note: Auto-assigned LCD_SERIAL_PORT. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
also /#define SERIAL_PORT_3 2  is not valid config line add the missing preceeding / 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-09-24 by ellensp

gsuresh2u, 2023-09-24:
Thanks for reply
modified code as 
```
#define SERIAL_PORT -1
#define LCD_SERIAL_PORT 1 
```
and disabled Serial3 port
Still same problem, screen is just Flickkering

gsuresh2u, 2023-09-24:
Solved my problem, its actually ribbon cable problem
Now its working fine

[image]