MarlinFirmware/Marlin #26307: [BUG] Ignore TMC_HOME_PHASE for core axes even if user configures phases

vlsi, 2023-09-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Core machines can't reliably settle on the same phase for homing core axes. For instance, with CoreXY, X and Y positions depend on both A and B motors, so the homing A phase depends on exact location of X and Y (== on the position of B motor).

### Expected behavior

I suggest adding checks to `backout_to_tmc_homing_phase` and return in case.

What do you think of the following?

```diff
@@ -1837,7 +1837,7 @@ void prepare_line_to_destination() {
       #define PHASE_PER_MICROSTEP(N) (256 / _MAX(1, N##_MICROSTEPS))

       switch (axis) {
-        #ifdef X_MICROSTEPS
+        #ifdef X_MICROSTEPS && NONE(CORE_IS_XY, CORE_IS_XZ)
           case X_AXIS:
             phasePerUStep = PHASE_PER_MICROSTEP(X);
             phaseCurrent = stepperX.get_microstep_counter();
@@ -1845,7 +1845,7 @@ void prepare_line_to_destination() {
             stepperBackoutDir = IF_DISABLED(INVERT_X_DIR, -)effectorBackoutDir;
             break;
         #endif
-        #ifdef Y_MICROSTEPS
+        #ifdef Y_MICROSTEPS && NONE(CORE_IS_XY, CORE_IS_YZ)
           case Y_AXIS:
             phasePerUStep = PHASE_PER_MICROSTEP(Y);
             phaseCurrent = stepperY.get_microstep_counter();
@@ -1853,7 +1853,7 @@ void prepare_line_to_destination() {
             stepperBackoutDir = IF_DISABLED(INVERT_Y_DIR, -)effectorBackoutDir;
             break;
         #endif
-        #ifdef Z_MICROSTEPS
+        #ifdef Z_MICROSTEPS && NONE(CORE_IS_XZ, CORE_IS_YZ)
           case Z_AXIS:
             phasePerUStep = PHASE_PER_MICROSTEP(Z);
             phaseCurrent = stepperZ.get_microstep_counter();
```

### Actual behavior

Marlin attempts to reach home_phase even for core machines

### Version of Marlin Firmware

2.1.2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://github.com/vlsi/reborn2-marlin/blob/vlsi-2.1.x/Marlin/Configuration.h

thisiskeithb, 2023-09-26:
Instead of ignoring the setting, it'd be better to add a compile-time check to `SanityCheck.h` for `TMC_HOME_PHASE` & `CORE*` / `MARKFORGED_*` kinematics.

Something like:

```cpp
#if ENABLED(TMC_HOME_PHASE)
  #if ENABLED(COREXY)
    #error "TMC_HOME_PHASE is incompatible with COREXY."
  #elif ENABLED(COREXZ)
    #error "TMC_HOME_PHASE is incompatible with COREXZ."
  #elif ENABLED(COREYZ)
    #error "TMC_HOME_PHASE is incompatible with COREYZ."
  #elif ENABLED(COREYX)
    #error "TMC_HOME_PHASE is incompatible with COREYX."
  #elif ENABLED(COREZX)
    #error "TMC_HOME_PHASE is incompatible with COREZX."
  #elif ENABLED(COREZY)
    #error "TMC_HOME_PHASE is incompatible with COREZY."
  #elif ENABLED(MARKFORGED_XY)
    #error "TMC_HOME_PHASE is incompatible with MARKFORGED_XY."
  #elif ENABLED(MARKFORGED_YX)
    #error "TMC_HOME_PHASE is incompatible with MARKFORGED_YX."
  #endif
#endif
```

vlsi, 2023-09-26:
TMC_HOME_PHASE for Z is compatible with COREXY, so the check should use static_assert or something like that.

At the same time, adding checks to backout_to_tmc_homing_phase would reduce the generated code size which is nice anyway.

vlsi, 2023-09-26:
The diff is `64` bytes flash for skpping `case X_AXIS` and `case Y_AXIS` in CoreXY, so adding checks to `set_axis_never_homed` might be omitted.

WDYT of the following?

```c
#ifdef TMC_HOME_PHASE
  #if ANY(IS_CORE, MARKFORGED_XY, MARKFORGED_YX)
    constexpr float phases[] = TMC_HOME_PHASE;
    static_assert(COUNT(phases) == NUM_AXES, "TMC_HOME_PHASE must have " _NUM_AXES_STR " elements (and no others).");
    NUM_AXIS_CODE(
      static_assert(phases[X_AXIS] == -1 || NONE(CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY, MARKFORGED_YX), "TMC_HOME_PHASE.X is incompatible with CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY, MARKFORGED_YX. Set TMC_HOME_PHASE.X to -1"),
      static_assert(phases[Y_AXIS] == -1 || NONE(CORE_IS_XY, CORE_IS_YZ, MARKFORGED_XY, MARKFORGED_YX), "TMC_HOME_PHASE.Y is incompatible with CORE_IS_XY, CORE_IS_YZ, MARKFORGED_XY, MARKFORGED_YX. Set TMC_HOME_PHASE.Y to -1"),
      static_assert(phases[Z_AXIS] == -1 || NONE(CORE_IS_XZ, CORE_IS_YZ), "TMC_HOME_PHASE.Z is incompatible with CORE_IS_XZ, CORE_IS_YZ. Set TMC_HOME_PHASE.Z to -1")
    );
    NUM_AXIS_CODE(
      static_assert(phases[X_AXIS] >= -1 || phases[X_AXIS] <= 1023, "TMC_HOME_PHASE.X must be between -1 and 1023"),
      static_assert(phases[Y_AXIS] >= -1 || phases[Y_AXIS] <= 1023, "TMC_HOME_PHASE.Y must be between -1 and 1023"),
      static_assert(phases[Z_AXIS] >= -1 || phases[Z_AXIS] <= 1023, "TMC_HOME_PHASE.Z must be between -1 and 1023"),
      static_assert(phases[I_AXIS] >= -1 || phases[I_AXIS] <= 1023, "TMC_HOME_PHASE.I must be between -1 and 1023"),
      static_assert(phases[J_AXIS] >= -1 || phases[J_AXIS] <= 1023, "TMC_HOME_PHASE.J must be between -1 and 1023"),
      static_assert(phases[K_AXIS] >= -1 || phases[K_AXIS] <= 1023, "TMC_HOME_PHASE.K must be between -1 and 1023"),
      static_assert(phases[U_AXIS] >= -1 || phases[U_AXIS] <= 1023, "TMC_HOME_PHASE.U must be between -1 and 1023"),
      static_assert(phases[V_AXIS] >= -1 || phases[V_AXIS] <= 1023, "TMC_HOME_PHASE.V must be between -1 and 1023"),
      static_assert(phases[W_AXIS] >= -1 || phases[W_AXIS] <= 1023, "TMC_HOME_PHASE.W must be between -1 and 1023")
    );
 #endif
#endif
```

thisiskeithb, 2023-09-26:
Closing since you've opened a PR.

closed 2023-09-26 by thisiskeithb