MarlinFirmware/Marlin #26309: [BUG] (bug summary)The printer does not start printing from the SD card, configuration with #define DGUS_LCD_UI E3S1PRO

Gorien, 2023-09-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If you enable #define DGUS_LCD_UI E3S1PRO, then nothing happens when you select a file from the SD card. The connected program (Pronterface) reports in the port monitor that a file with the same name was not found. During the experiments, it was revealed that the problem is with long names and if in the file: Marlin\src\lcd\extui\dgus_e3s1pro\DGUSSDCardHandler_Advanced.cpp on line 73 setFilename(currentFile, fileList.filename(), fileList.isDir()); replace the ui-api function: fileList.filename() with fileList.shortFilename(), then everything works, but only DOS names are displayed in the display menu. The #define LONG_FILENAME_HOST_SUPPORT option does not help.

### Version of Marlin Firmware

2.1.3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-09-26:
Please submit PR if you got it working.
LONG_FILENAME_HOST_SUPPORT is irrelevant to DGUS screens.

Gorien, 2023-09-26:
> Please submit PR if you got it working. LONG_FILENAME_HOST_SUPPORT is irrelevant to DGUS screens.

what is PR?

EvilGremlin, 2023-09-26:
pull request

Gorien, 2023-09-26:
> Please submit PR if you got it working. LONG_FILENAME_HOST_SUPPORT is irrelevant to DGUS screens.

thisiskeithb, 2023-09-26:
Did this end up not being a bug?

EvilGremlin, 2023-09-26:
i think we'll see fix PR soon

Gorien, 2023-09-26:
> Did this end up not being a bug?

I'm not a programmer, I'm just trying to build firmware for my Creality Sermoon V1. I think this is a bug, because printing from the SD card does not start, something is wrong with long names.

ellensp, 2023-09-26:
Please attach your configuration files.

Gorien, 2023-09-26:
> Please attach your configuration files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/12725629/Marlin.zip)

thinkyhead, 2023-10-07:
It appears that the code submitted by @strnk in #25905 is incomplete. This displays the selected filename but never selects and starts it for printing. The original author may have a fix in the works. If not, we should try to fill in the missing pieces ourselves.
```cpp
void DGUSScreenHandler::startPrintFromSD(const char* const filename) {
  triggerScreenChange(DGUS_ScreenID::HOME);
  sdPrintFilename = filename;
  setStatusMessage(sdPrintFilename, 0);
}
```

emuell36, 2023-10-20:
Could yo give me a hint where to add (?) this code?
I am completly new to this whole "business" and a little bit lost :/

strnk, 2023-10-20:
@emuell36 the code is the one currently in the repo, which needs to be fixed.

I'm quite sure I either missed a commit somewhere during my PR, or the job is handled somewhere else, because I did print from SD on my Ender 3 S1+ using this.

I can give it a look in a couple weeks, I don't have access to my computer until then.

closed 2024-02-02 by github-actions[bot]