MarlinFirmware/Marlin #26312: [BUG] M420 V1 T0 return invalid mesh after a M502

gabin1369, 2023-09-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,
I have just noticed that if I send M502 while a mesh is valid, I receive an invalid mesh message after sending M502. I have tested it with different motherboards and Marlin versions.

### Bug Timeline

probably since the first version

### Expected behavior

I expected the same result as before sending M502

### Actual behavior

If I level the bed with G29 and save the result M500, M420 V1 T0 return a valid mesh. I reset settings to their factory defaults with a M502 and the new answer to  M420 V1 T0 is "invalid mesh". Thanksfully, if I reboot the printer, the new answer to M420 V1 T0 is a valid mesh as you can see:

> >>> m420 v1 t0
SENDING:M420 V1 T0
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.105 -0.191 -0.299 -0.137 +0.126
 1 +0.121 +0.130 -0.011 +0.115 +0.115
 2 +0.045 +0.104 +0.061 +0.109 +0.046
 3 +0.095 +0.152 +0.208 +0.096 +0.064
 4 +0.305 +0.543 +0.561 +0.325 +0.105
echo:Bed Leveling OFF
echo:Fade Height 5.00
ok P31 B15
>>> m500
SENDING:M500
echo:Settings Stored (756 bytes; crc 20190)
//action:notification Config. enregistree
//action:notification Config. enregistree
ok P31 B15
>>> m420 v1 t0
SENDING:M420 V1 T0
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.105 -0.191 -0.299 -0.137 +0.126
 1 +0.121 +0.130 -0.011 +0.115 +0.115
 2 +0.045 +0.104 +0.061 +0.109 +0.046
 3 +0.095 +0.152 +0.208 +0.096 +0.064
 4 +0.305 +0.543 +0.561 +0.325 +0.105
echo:Bed Leveling OFF
echo:Fade Height 5.00
ok P31 B15
>>> m502
SENDING:M502
//action:notification Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
==Dwin Init Complete==
ok P31 B15
>>> m420 v1 t0
SENDING:M420 V1 T0
echo:Invalid mesh.
echo:Bed Leveling OFF
echo:Fade Height 5.00
ok P31 B15
Disconnected.
////////////////////////
//// PRINTER REBOOT ////
////////////////////////
Connecting...
Printer is now online.
>>> m420 v1 t0
SENDING:M420 V1 T0
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.105 -0.191 -0.299 -0.137 +0.126
 1 +0.121 +0.130 -0.011 +0.115 +0.115
 2 +0.045 +0.104 +0.061 +0.109 +0.046
 3 +0.095 +0.152 +0.208 +0.096 +0.064
 4 +0.305 +0.543 +0.561 +0.325 +0.105
echo:Bed Leveling ON
echo:Fade Height 5.00
ok P31 B15

I join an example of config files with a gcode file that create a mesh (mesh.gcode)

### Steps to Reproduce

1. launch a mesh leveling with a G29
2. save the result with M500
3. check mesh with M420 V1 T0
4. reset settings to their factory defaults with M502
5. check mesh with M420 V1 T0

### Version of Marlin Firmware

Marlin 2.1.2.1 with bug fixes

### Printer model

Creality Ender7

### Electronics

stock 3d printer 

### Bed Leveling

ABL Linear grid

### Your Slicer

IdeaMaker

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[M502bug.zip](https://github.com/MarlinFirmware/Marlin/files/12740764/M502bug.zip)

ellensp, 2023-09-27:
M502 is meant to reset all setting to firmware defaults, default is no mesh set. This is doing what it should

gabin1369, 2023-09-27:
Thank you for your quick answer. 
You’re absolutely right, I should have sent M501 instead of M502. My bad
Have a good day

closed 2023-09-27 by ellensp