MarlinFirmware/Marlin #26313: [BUG] Setting EXTRUDERS to 5 & MMU_MODEL to PRUSA_MMU2S disables X Axis on SKR 3 EZ

eoyilmaz, 2023-09-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Setting `EXTRUDERS` to `5` and `MMU_MODEL` to `PRUSA_MMU2S` (and all the other required setting for MMU2s to work) disables X-Axis movement on SKR 3 EZ board. I tested it with SKR Mini E3 v3 and it was fine there. Did a bisect but this newer worked since the introduction of the SKR 3 (EZ) board.

### Bug Timeline

Always present

### Expected behavior

X-Axis to move

### Actual behavior

X-Axis doesn't move.

### Steps to Reproduce

1. Set `EXTRUDEES` to `5`
2. Set `MMU_MODEL` to `PRUSA_MMU2S`
3. Build the firmware and install it
4. Move X-Axis from the menu

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Sep 27 2023 17:40:51) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro SKR 3 EZ MMU2s EXTRUDER_COUNT:5

### Printer model

Creality Ender3 Pro

### Electronics

SKR 3 EZ

### Add-ons

CRTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Here is the minimal config to reproduce this:

[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/12741528/Marlin_Config.zip)

thinkyhead, 2023-10-06:
It sounds like a potential pin conflict. Please enable `PINS_DEBUGGING` and issue an `M43` to get a pins report, then look for any conflicts between pins. That might provide the clue we need to solve the issue.

eoyilmaz, 2023-10-12:
Okay, that makes sense, but it will take some time for me to try it as I had to return back to an SKR Mini E3 v3 board to finish some work...

eoyilmaz, 2023-11-23:
Sorry for my late response, I had to finish some work and come back to this board.

So here is the M43 command output:

```
Send: M43
Recv: PIN: PA0  (A0)   M42 P0           E1_DIAG_PIN                            Input  LOW
Recv: PIN: PA1  (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA2  (A2)   M42 P2           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA3  (A3)   M42 P3           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA4  (A4)   M42 P4           EXP2_04_PIN                            Input  LOW
Recv: .                                 SD_SS_PIN                              Input  LOW
Recv: PIN: PA5  (A5)   M42 P5           EXP2_02_PIN                            Input  LOW
Recv: .                                 DOGLCD_SCK                             Input  LOW
Recv: .                                 SD_SCK_PIN                             Input  LOW
Recv: PIN: PA6  (A6)   M42 P6           EXP2_01_PIN                            Input  LOW
Recv: .                                 SD_MISO_PIN                            Input  LOW
Recv: PIN: PA7  (A7)   M42 P7           EXP2_06_PIN                            Input  LOW
Recv: .                                 DOGLCD_MOSI                            Input  LOW
Recv: .                                 SD_MOSI_PIN                            Input  LOW
Recv: PIN: PA8         M42 P8           Y_DIR_PIN                              protected
Recv: PIN: PA9         M42 P9           <unused/unknown>                       Input  HIGH
Recv: PIN: PA10        M42 P10          <unused/unknown>                       Alt Function: 10 - OTG
Recv: PIN: PA11        M42 P11          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA12        M42 P12          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA13        M42 P13          I2C_SDA_PIN                            Input  HIGH
Recv: PIN: PA14        M42 P14          I2C_SCL_PIN                            Input  HIGH
Recv: PIN: PA15        M42 P15          Y_STEP_PIN                             protected
Recv: PIN: PB0  (A8)   M42 P16          BTN_ENC                                Input  HIGH
Recv: .                                 EXP1_02_PIN                            Input  HIGH
Recv: PIN: PB1  (A9)   M42 P17          BTN_EN1                                Input  HIGH
Recv: .                                 EXP1_03_PIN                            Input  HIGH
Recv: PIN: PB2         M42 P18          EXP2_05_PIN                            Input  LOW
Recv: PIN: PB3         M42 P19          HEATER_0_PIN                           protected
Recv: PIN: PB4         M42 P20          HEATER_1_PIN                           Output LOW
Recv: PIN: PB5         M42 P21          FAN0_PIN                               protected
Recv: PIN: PB6         M42 P22          FAN1_PIN                               protected
Recv: .                                 E0_AUTO_FAN_PIN                        protected
Recv: PIN: PB7         M42 P23          FAN2_PIN                               protected
Recv: PIN: PB8         M42 P24          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB9         M42 P25          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB10        M42 P26          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB11        M42 P27          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB12        M42 P28          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB13        M42 P29          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB14        M42 P30          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB15        M42 P31          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PC0  (A10)  M42 P18          Z_MIN_PIN                              protected
Recv: .                                 Z_STOP_PIN                             protected
Recv: .                                 Z_DIAG_PIN                             protected
Recv: PIN: PC1  (A11)  M42 P19          X_MIN_PIN                              protected
Recv: .                                 X_STOP_PIN                             protected
Recv: .                                 X_DIAG_PIN                             protected
Recv: PIN: PC2  (A14)  M42 P22          FIL_RUNOUT_PIN                         Input  HIGH
Recv: .                                 E0_DIAG_PIN                            Input  HIGH
Recv: PIN: PC3  (A15)  M42 P23          Y_MIN_PIN                              protected
Recv: .                                 Y_STOP_PIN                             protected
Recv: .                                 Y_DIAG_PIN                             protected
Recv: PIN: PC4  (A12)  M42 P20          EXP2_07_PIN                            Input  LOW
Recv: PIN: PC5  (A13)  M42 P21          BEEPER_PIN                             Output LOW
Recv: .                                 EXP1_01_PIN                            Output LOW
Recv: PIN: PC6         M42 P36          E0_SERIAL_TX_PIN                       Output HIGH
Recv: .                                 E0_SERIAL_RX_PIN                       Output HIGH
Recv: PIN: PC7         M42 P37          E0_ENABLE_PIN                          protected
[... cut ...]
Recv: PIN:  º\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN:  º\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN:  º\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN:  º\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN:  º\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN:  º\x05\x08Í (A T:22.10 /0.00 B:22.56 /0.00 @:0 B@:0
```

eoyilmaz, 2023-11-23:
And here is the output when `EXTRUDERS 1` and `MMU_MODEL` is commented out:

```
Recv: PIN: PA0  (A0)   M42 P0           E1_DIAG_PIN                            Input  LOW
Recv: PIN: PA1  (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA2  (A2)   M42 P2           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA3  (A3)   M42 P3           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PA4  (A4)   M42 P4           EXP2_04_PIN                            Input  LOW
Recv: .                                 SD_SS_PIN                              Input  LOW
Recv: PIN: PA5  (A5)   M42 P5           EXP2_02_PIN                            Input  LOW
Recv: .                                 DOGLCD_SCK                             Input  LOW
Recv: .                                 SD_SCK_PIN                             Input  LOW
Recv: PIN: PA6  (A6)   M42 P6           EXP2_01_PIN                            Input  LOW
Recv: .                                 SD_MISO_PIN                            Input  LOW
Recv: PIN: PA7  (A7)   M42 P7           EXP2_06_PIN                            Input  LOW
Recv: .                                 DOGLCD_MOSI                            Input  LOW
Recv: .                                 SD_MOSI_PIN                            Input  LOW
Recv: PIN: PA8         M42 P8           Y_DIR_PIN                              protected
Recv: PIN: PA9         M42 P9           <unused/unknown>                       Input  HIGH
Recv: PIN: PA10        M42 P10          <unused/unknown>                       Alt Function: 10 - OTG
Recv: PIN: PA11        M42 P11          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA12        M42 P12          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA13        M42 P13          I2C_SDA_PIN                            Input  HIGH
Recv: PIN: PA14        M42 P14          I2C_SCL_PIN                            Input  HIGH
Recv: PIN: PA15        M42 P15          Y_STEP_PIN                             protected
Recv: PIN: PB0  (A8)   M42 P16          BTN_ENC                                Input  HIGH
Recv: .                                 EXP1_02_PIN                            Input  HIGH
Recv: PIN: PB1  (A9)   M42 P17          BTN_EN1                                Input  HIGH
Recv: .                                 EXP1_03_PIN                            Input  HIGH
Recv: PIN: PB2         M42 P18          EXP2_05_PIN                            Input  LOW
Recv: PIN: PB3         M42 P19          HEATER_0_PIN                           protected
Recv: PIN: PB4         M42 P20          HEATER_1_PIN                           Output LOW
Recv: PIN: PB5         M42 P21          FAN0_PIN                               protected
Recv: PIN: PB6         M42 P22          FAN1_PIN                               protected
Recv: .                                 E0_AUTO_FAN_PIN                        protected
Recv: PIN: PB7         M42 P23          FAN2_PIN                               protected
Recv: PIN: PB8         M42 P24          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB9         M42 P25          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB10        M42 P26          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB11        M42 P27          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB12        M42 P28          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB13        M42 P29          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB14        M42 P30          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PB15        M42 P31          <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: PC0  (A10)  M42 P18          Z_MIN_PIN                              protected
Recv: .                                 Z_STOP_PIN                             protected
Recv: .                                 Z_DIAG_PIN                             protected
Recv: PIN: PC1  (A11)  M42 P19          X_MIN_PIN                              protected
Recv: .                                 X_STOP_PIN                             protected
Recv: .                                 X_DIAG_PIN                             protected
Recv: PIN: PC2  (A14)  M42 P22          FIL_RUNOUT_PIN                         Input  HIGH
Recv: .                                 E0_DIAG_PIN                            Input  HIGH
Recv: PIN: PC3  (A15)  M42 P23          Y_MIN_PIN                              protected
Recv: .                                 Y_STOP_PIN                             protected
Recv: .                                 Y_DIAG_PIN                             protected
Recv: PIN: PC4  (A12)  M42 P20          EXP2_07_PIN                            Input  LOW
Recv: PIN: PC5  (A13)  M42 P21          BEEPER_PIN                             Output LOW
Recv: .                                 EXP1_01_PIN                            Output LOW
Recv: PIN: PC6         M42 P36          E0_SERIAL_TX_PIN                       Output HIGH
Recv: .                                 E0_SERIAL_RX_PIN                       Output HIGH
Recv: PIN: PC7         M42 P37          E0_ENABLE_PIN                          protected
Recv: PIN: PC8         M42 P38          <unused/unknown>                       Alt Function: 12 - FSMC, SDIO, OTG
[... cut ...]
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: PIN: Êy\x05\x08Í (A1)   M42 P1           <unused/unknown>   Analog in =     0   Input  LOW
Recv: ok
```

eoyilmaz, 2023-11-23:
I had to update both of my comments a couple of times, as OctoPrint was not outputting them correctly, but I think it is now okay.

eoyilmaz, 2023-11-23:
I think this is related with the `MMU2_SERIAL_PORT` setting, and setting it to 2 disables the X axis where as setting it to 1 fixes the X axis and the communication with MMU.

So, this is solved and we can close this ticket.

Thank you!

closed 2023-11-23 by eoyilmaz