MarlinFirmware/Marlin #26316: [BUG] tft_string.cpp compilation error

alexgrach, 2023-09-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin\src\lcd\tft\tft_string.cpp:132:29: error: 'struct unifont_t' has no member named 'Format'; did you mean 'format'?

### Version of Marlin Firmware

2.1.3

### Printer model

Kingroon KP3S

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12752971/Configuration.zip)
[0001-Update-tft_string.cpp.patch](https://github.com/MarlinFirmware/Marlin/files/12752977/0001-Update-tft_string.cpp.patch)

EvilGremlin, 2023-09-28:
already https://github.com/MarlinFirmware/Marlin/pull/26292

closed 2023-09-28 by thisiskeithb

thinkyhead, 2023-10-06:
Duplicate of https://github.com/MarlinFirmware/Marlin/pull/26292