MarlinFirmware/Marlin #26317: [BUG] IDEX, Duplication mode, Dual Filament Sensor, Second Sensor being ignored.

gaby64, 2023-09-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I installed two BTT SFS 2.0 on my IDEX printer, these are smart filament sensors with encoder wheel and filament switch.
When printing in Duplication mode, the second SFS ignores the encoder wheel and only registers runout from the filament switch and this does not trigger the filament change for that extruder, rather only the main extruder.

### Expected behavior

printer should trigger filament change on the extruder for which the sensor matches and both the filament switch and the encoder wheel should be monitored, regardless of IDEX mode, it should be based on extruder activity.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

JGMaker Artist-D Pro

### Electronics

Robin Pro v1

### Add-ons

BTT SFS 2.0

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_JGMADP.zip](https://github.com/MarlinFirmware/Marlin/files/12754532/Marlin_JGMADP.zip)

gaby64, 2024-01-13:
Really?

thisiskeithb, 2024-01-13:
> Really?

As the bot said:

> This issue has had no activity for the last 90 days.
> Do you still see this issue with the latest bugfix-2.1.x code?
> **Please add a reply within 14 days or this issue will be automatically closed.**

I’ll reopen the issue, but you’ll need to follow the bot’s instructions next time and test the latest code & add a reply until someone can look at this / confirm the bug.

Also:

> Disclaimer: This is an open community project with lots of activity and limited resources. The main project contributors will do a bug sweep ahead of the next release, but any skilled member of the [...]

gaby64, 2024-04-27:
I still see the issue

ellensp, 2024-06-27:
I have updated the Configs to current bugfix 
[Updated Configuration.zip](https://github.com/user-attachments/files/16009137/Updated.Configuration.zip)

also needs 
```DIFF
diff --git a/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_PRO.h b/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_PRO.h
index d5689c1e71..6577cab140 100644
--- a/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_PRO.h
+++ b/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_PRO.h
@@ -191,7 +191,9 @@
 //#define TEMP_0_CS_PIN                     PE5   // TC1 - CS1
 //#define TEMP_0_CS_PIN                     PF11  // TC2 - CS2
 
-#define POWER_LOSS_PIN                      PA2   // PW_DET
+#ifndef POWER_LOSS_PIN
+  #define POWER_LOSS_PIN                      PA2   // PW_DET
+#endif
 #define FIL_RUNOUT_PIN                      PA4   // MT_DET1
 #define FIL_RUNOUT2_PIN                     PE6   // MT_DET2
 #define FIL_RUNOUT3_PIN                     PG14  // MT_DET3
```

gaby64, 2024-07-09:
Flashed it, issue is still present.

gaby64, 2024-10-09:
I do

gaby64, 2025-01-18:
still got the issue

closed 2025-05-03 by github-actions[bot]