MarlinFirmware/Marlin #26319: [BUG] (bug summary)

anotherAdub, 2023-10-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Incorrect Menu Display when Pausing a a Job. The display changes to Resume, but does not toggle the display when clicked.

### Bug Timeline

First time I used pause job

### Actual behavior

When Clicking the Menu it should change to say Pause or Resume. 

### Steps to Reproduce

Pause the Job

Below it asks about bed levelling - I do not know what type! I just chose one.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Tronky XY Pro-2

### Electronics

STock  Chitu3DV6 Baud 115200 protocol 1.0 PSU generic

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No Config (how do I get it?) as this is a firmware issue anyway.
[files.zip](https://github.com/MarlinFirmware/Marlin/files/12777562/files.zip)

ellensp, 2023-10-02:
No real configs, no way to reproduce or even look at issue

No such printer as Tronky XY Pro-2, there are several TRONXY XY-2 PRO printers.

closed 2023-10-02 by ellensp

thinkyhead, 2023-10-06:
> No Config (how do I get it?) as this is a firmware issue anyway.

First you have to get the latest Marlin (bugfix-2.1.x). Then get the configurations for your machine from our examples repository so you can test the latest Marlin code. If there is no example for your machine then it is not yet supported by mainline Marlin, and you will need to get your support from Tronxy.