MarlinFirmware/Marlin #26321: [FR] Support generic 2004 I2C LCD

JohnTheCoolingFan, 2023-10-02:
### Is your feature request related to a problem? Please describe.

I want to use a generic I2C HD44780 20x4 character LCD module.

### Are you looking for hardware support?

It's a module similar to 16x2 lcd, which is supported via LCM1602 feature I believe. It uses the same controller and an i2c port expander. Here's the image of one: 
[image]

### Describe the feature you want

I want Marlin to support using a generic lcd 20x4 module that works over I2C. Other configurations that do use this module don't suit my usecase I think.

### Additional context

I'm sorry if this is suplicate and/or this is already implemented, I've tried searching through issues, code and documentation but couldn't find any conclusive information.

ellensp, 2023-10-02:
This should work.

#define LCM1602
#define HAS_WIRED_LCD 1
#define HAS_MARLINUI_HD44780 1
#define LCD_WIDTH 20
#define LCD_HEIGHT 4

not that is much use without a encoder

JohnTheCoolingFan, 2023-10-02:
Looks like this would work. But I don't see those options in `Configuration.h` or `Configuration_adv.h` or docs on the website. I'm confused on why these options are not documented.

I'll use an encoder for the build I'm planning (heavy modification of a bought printer that uses Marlin)

I'll close this issue without testing because I don't have time to test it now.

Thanks!

closed 2023-10-02 by JohnTheCoolingFan

thinkyhead, 2023-10-07:
@ellensp — Are those extra settings required but not being set in `Conditionals_LCD.h`?
```cpp
#define HAS_WIRED_LCD 1
#define HAS_MARLINUI_HD44780 1
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
```

ellensp, 2023-10-07:
there is no LCM1602 section in Conditionals_LCD.h