MarlinFirmware/Marlin #26322: [FR] Serial reading via DMA to improve serial communication reliability

rondlh, 2023-10-03:
### Is your feature request related to a problem? Please describe.

This feature requests is related to unreliable serial data communication I found when putting load on the motherboard by printing faster. I use a TFT with serial interface to print. I found that under high loads the motherboard (MKS Monster8 V1.0) sometimes drops a character. I managed to confirm this by logging received commands in Marlin and in parallel monitor the data send to the motherboard by an independent PC serial adapter. By using a checksum I get a trigger when Marlin finds an issue, so then I can compare what data Marlin received and what the PC adapter found. I consistently see that Marlin "drops" one character once in a while. I never see any data corruption like a changed character or such, only a missing character. So I suspect that sometimes the interrupt on my motherboards (STM32F407) sometimes cannot keep up with the data.

A possible way to solve this is by using DMA to read the data in the background to a cyclical read buffer. This process runs completely independent once started, not even a interrupt service routine is needed. Functions like "serial.available" can check the DMA register to see how much data has become available and update the head index of the buffer, then the data can be processed. A strategy like this is successfully implemented in the BTT TFT Firmware: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Marlin uses Arduino STM32 for it's serial communication, and there I get stuck on how to test/implement this.
I have experience in working with DMA controllers on STM32F1/2/4 and GD32, I can support that part.

### Are you looking for hardware support?

No

### Describe the feature you want

DMA serial reading to improve serial reading reliability

### Additional context

I have experience in working with DMA controllers on STM32F1/2/4 and GD32, I can support that part.
Additionally to implement DMA serial writing is quite trivial, but that is not very critical.

EvilGremlin, 2023-10-03:
You have to write this in STM32 HAL (MarlinSerial::begin i guess) and make it completely transparent for all other codebase. While some small arch-dependent branching is ok in main code, it should be kept to absolute minimum.

rondlh, 2023-10-03:
> You have to write this in STM32 HAL (starting with MarlinSerial::begin i guess) and make it completely transparent for all other codebase. While some small arch-dependent branching is ok in main code, it should be kept to absolute minimum.

I have a bare metal implementation starting at "nothing" and handling the channel/stream selection depending on the USART used, ending with Serial_Get and Serial_Put. Using STM32 HAL would be very easy.

EvilGremlin, 2023-10-03:
It'd be really cool if you can replace another piece of junk arduino code.
btw, you don't have to bother with flow control, it does not benefit our text terminal case. You can dig up thread on octoprint forum where this was attempted and failed.

rondlh, 2023-10-04:
I see what needs to happen, basically a HardwareSerial class (HardwareSerial.cpp/h, uart.c/h) needs to be made. Lots of things in there are not even needed.

One odd thing I noticed, when I have data corruption (dropped character) there is a call to HAL_UART_ErrorCallback in uart.c, but ErrorCode is 0, no error flags are set, HAL_UART_GetError(huart) reports a 0. What's going on?

rondlh, 2023-10-07:
I managed to isolate Class HardwareSerial out of Arduino so I can modify things.
Then created a class HardwareSerial2, and updated Marlin to work with HardwareSerial2 (just rename).
I disabled EMERGENCY_PARSER for now, but I can get that working too if needed.
I don't know how to overwrite a method of HardwareSerial, only a few changes are required. Anybody???

**Changed methods:**
void HardwareSerial::begin(unsigned long baud, uint8_t config) (Start the free running DMA process)
int HardwareSerial2::available(void) (Update the buffer head pointer to the current DMA progress)
int HardwareSerial2::peek(void) (not sure if it is even used) (Update the buffer head pointer to the current DMA progress)
int HardwareSerial2::read(void) (Update the buffer head pointer to the current DMA progress)

Then it works already... and the best thing is, it solves my data corruption issue. I think the issue is actually in the Arduino platform, but it only shows under stress. Now 250K baud is stable. I'm going to test faster serial speeds now.
I will release the source code after a cleanup (STM32F2-F4 only). Anybody interested?

EvilGremlin, 2023-10-07:
Yep, just open draft PR. We can clean it up and integrate properly later. Better name it same HardwareSerial and wrap in #if ENABLED(SERIAL_DMA), something like that.

thisiskeithb, 2023-10-07:
> I disabled EMERGENCY_PARSER for now, but I can get that working too if needed.

It'd be good to get `EMERGENCY_PARSER` working.

EvilGremlin, 2023-10-07:
Yep, EMERGENCY_PARSER is pretty much obligatory by now

rondlh, 2023-10-07:
OK OK OK, no problem, EMERGENCY_PARSER IS WORKING... took a whole 4 lines of additional code.
It seems the EMERGENCY_PARSER just gets all available data in the RX buffer immediately. 

I had to cheat a little bit to get it working, with interrupt based serial reading (1 interrupt per received character), the received data is just stored in the RX buffer and forwarded to EMERGENCY_PARSE. But here I use serial DMA to a circular RX buffer, so I don't get an interrupt for every byte received, I don't even use an interrupt handler. I could enable the Serial Idle interrupt, which would generate an interrupt after a message is received IF, and only if, the serial line goes to idle afterwards, which will not happening if commands keep coming (like with repeated M118 commands). So what I did instead is that when Marlin check for available data, or reads or peeks data, then I update the RX Buffer head with the current reported DMA status, and send all new data to the Emergency parse. I gave it a quick try when running my printer at 10%, and it seems to work fine.

I did some more speed testing, before I would have to lower the serial speed to 115200 baud to make sure the communication is stable, with DMA reading enabled I can run the printer at 1M baud without any issues. I know the communication is correct , at least towards the motherboards, because I use a checksum. I also use a stress test that sends over 1400 messages/s to the motherboards and get an Advanced OK back and the response to the send M118 command. I'm sure there is something going wrong in the Arduino framework!
[image]

I forked Marlin, please note that this is very experimental code, and it is likely to make your printer explode. I tested it on an MKS Monster8 V1.0 (STM32F407), the code should "work" on STM32F2/F4 MCU, if you use USART1-4.

EvilGremlin, 2023-10-07:
Are F1/F7/H7 series DMA different? or you didn't look at it?

rondlh, 2023-10-07:
> Are F1/F7/H7 series DMA different? or you didn't look at it?

F1 is a little bit different, some register names are different (see below).
I'm not sure about F7 and H7, but the actual code needed to get this working is not so much. 
So if you want it and can read a STM32 Reference Manual, then you should be able to get it working in an hour or 2.

F1:
// CCR    DMA stream x configuration register
// CNDTR  DMA stream x number of data register
// CPAR   DMA stream x peripheral address register
// CMAR   DMA stream x memory 0 address register
//
// ISR    DMA low/high interrupt status register
// IFCR   DMA low/high interrupt flag clear register

F2/F4:
// CR     DMA stream x configuration register
// NDTR   DMA stream x number of data register
// PAR    DMA stream x peripheral address register
// M0AR   DMA stream x memory 0 address register
//
// LISR   DMA low interrupt status register
// HISR   DMA high interrupt status register
// LIFCR  DMA low interrupt flag clear register
// HIFCR  DMA high interrupt flag clear register

rondlh, 2023-10-07:
PR is up, but here also the files to get this going. Drop in "src\HAL\STM32", overwrite MarlinSerial.h/c, added 2 new files, HardwareSerial2.h/c
[src-HAL-STM32-MarlinSerial.zip](https://github.com/MarlinFirmware/Marlin/files/12838980/src-HAL-STM32-MarlinSerial.zip)

Good luck!

BTW: Adding DMA serial writing is also possible, but doesn't bring much in my view. Interrupt writing is fine, but interrupt serial reading (at least Arduino's version) is not.

EvilGremlin, 2023-10-07:
no PR
[image]

rondlh, 2023-10-08:
@EvilGremlin Now it's there... #26328

rondlh, 2023-10-10:
I can wrap the code with "#if ENABLED(SERIAL_DMA)", no problem.

How important is F7 and H7 support? There are only a few boards like this around. 

A key problem I currently have is that a lot of build test get cancelled, and a few fail because "SERIAL_PORT 0", if I change that to 1 then they also build correctly. Anybody know how to deal with this?

EvilGremlin, 2023-10-10:
Wrong topic. I guess CI can't pull configs, again. You can only wait.

thisiskeithb, 2023-10-28:
Closing since you’ve opened a PR.

PR #26328 (2023-12-12): DMA Serial Reading to Circular buffer
VERY EXPERIMENTAL BUILD THAT USES DMA CIRCULAR READING TO THE RX SERIAL BUFFER, NO ISR NEEDED.

The Dma Controller Just Does Its Work In The Background And Keeps Copying The Received Serial Characters To The Rx Buffer.
Marlin Will Read And Process The Received Data, Freeing Up The Buffer.
Emergency Parser Is Working Too, But A Bit Different Than Before.

By Default Marlin Uses Interrupt Based Serial Reading Provided By The Arduino Platform.
On My Setup This Causes Data Corruption (Lost Bytes) On My Setup Starting At 250k Baud When Stressing The System.
In  [... cut]
closed 2023-10-28 by thisiskeithb