MarlinFirmware/Marlin #26323: [BUG] Z height isn't changing first couple of layers

KIWIshot, 2023-10-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I have a problem when I start printing Z height change seems to be ignored for first couple of layers after like 5th layer it starts to move but layers are already overoverlapped at his point. It started after I changed some settings in Marlyn and ressets Eprom memory.
When I returned the setting problem remained.  

### Bug Timeline

it started a couple of months ago but I haven't had time to look at it

### Expected behavior

I expect it to change height as it is in Gcode 

### Actual behavior

It isn't moving for the first 5 layers 

### Version of Marlin Firmware

 bugfix-2.1.x

### Printer model

Anet a8

### Electronics

SKR mini E3 V3

### Add-ons

induction sensor for Z axis

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12805077/Configuration.zip)

ellensp, 2023-10-04:
"When I say it to move in terminal it does but when Gcode is running it isn't.

Then there is something wrong with your gcode,

KIWIshot, 2023-10-05:
> "When I say it to move in terminal it does but when Gcode is running it isn't.
> 
> Then there is something wrong with your gcode,

Do you have any idea what could be wrong I have tried to make simple code with changing z height at different speeds and the same thing happened ➡ first five were ignored and then it started to move up

classicrocker883, 2023-10-05:
sounds more like a hardware issue. check your Z axis make sure it's moving smoothly. decouple the gantry from the motor, and the axis should move freely. it should not bind or feel like a flat spot moving it up and down by hand. also if it's a threaded rod check there isn't any issue with that. you may need to grease/lube it up.

thinkyhead, 2023-10-06:
Check to see if there is any difference between movement after enabling leveling with `M420 S1 Z0` (no fade) versus `M420 S1 Z5` (5mm fade). Also check your stored mesh with `M420 V` to make sure it centers nicely around zero. If it doesn't center around zero this can cause some unusual behavior in conjunction with leveling Z fade.

closed 2024-09-23 by github-actions[bot]