MarlinFirmware/Marlin #26324: [FR] Set #define Z_MAX_POS via LCD

Heinz-Loepmeier, 2023-10-05:
### Describe the feature you want

I'm using a hardware Z-max endstop for homing my Z-axis.
Currently I have no way of ajusting max Z position/ max Z travel.
Best I can do is using Homing offsets via LCD, but that only works in one direction. (Can't enter negative values, connecting via host and entering negative values into eeprom work, but I love to do that via LCD)

Other firmwares use 'Set current position as Z0.0' function, that works as well.

### Additional context

I'm using Marlin 2.1

classicrocker883, 2023-10-05:
what printer is this for? and what LCD, what UI, if you compile your own firmware can you post your config files, or link to any examples?

Heinz-Loepmeier, 2023-10-06:
> what printer is this for? and what LCD, what UI, if you compile your own firmware can you post your config files, or link to any examples?
i3 style printer with max Z enstop + 'RepRapDiscount Full Graphic Smart Controller' LCD.  
Here are the configs: https://github.com/Heinz-Loepmeier/Marlin-2.1-MKS-Gen-L-2.1  

Looks like @thinkyhead already thought about the same thing, but forgot about us zmax endstop users. 
https://github.com/MarlinFirmware/Marlin/issues/863

thisiskeithb, 2024-09-06:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/25582

closed 2024-09-06 by thisiskeithb