MarlinFirmware/Marlin #26333: Request for Assistance: Adding 2 New Axes and Replacing Extruder Ports E0 and E1 in Marlin for a 5-Axis CNC

Eng-Hajiarab, 2023-10-08:
### Describe the feature you want

Subject: Request for Assistance: Adding 2 New Axes and Replacing Extruder Ports E0 and E1 in Marlin for a 5-Axis CNC

Dear Marlin Development Team,

I hope this message finds you well. I am writing to seek your guidance and support in a project I'm working on that involves customizing Marlin firmware.

I have a specific requirement to add 2 new axes to my 5-axis CNC setup and replace the extruder ports E0 and E1 on the RAMPS board. My hardware configuration consists of an Arduino Mega 2560 and a RAMPS 1.4 board.

Here are the key changes I aim to make:

1. Add 2 New Axes for a 5-Axis CNC:
   - Configure Marlin to recognize and control a linear axis connected to port E0 using a TB6600 driver.
   - Configure Marlin to recognize and control a rotary axis connected to port E1 using another TB6600 driver.

2. Replace Extruder Ports E0 and E1:
   - Replace the existing extruder ports E0 and E1 with the newly added linear and rotary axes on the RAMPS board.

I want to emphasize that I do not intend to create a 3D printer; therefore, I do not require the extruder ports. The personal device I am creating is a 5-axis CNC machine.

I understand that this customization involves changes to the Marlin firmware, and I am eager to learn how to implement these modifications correctly.

Your expertise and guidance on how to configure Marlin for these changes would be greatly appreciated. Additionally, if there are any best practices or considerations I should be aware of while making these alterations, please do share them.

Thank you for your time and support. I look forward to your valuable insights and assistance.

Best regards,

Mahdi
Eng.hajiarab@gmail.com

ellensp, 2023-10-08:
This already exists

please see https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/linear_axes/RAMPS%205%20LINEAR_AXES

closed 2023-10-08 by ellensp

DerAndere1, 2023-11-03:
If you want to configure the rotary axis, you can test this pull request: https://github.com/MarlinFirmware/Marlin/pull/26174. In order to use tool centerpoint control you need to add inverse kinematics for your machine. I have added tool center point control (G43.4) for some 5 axis CNC machine geometries in my fork https://github.com/DerAndere1/Marlin , see my README file for details. You might have to change the implementation in one of these files: https://github.com/DerAndere1/Marlin/blob/Marlin2ForPipetBot/Marlin/src/module/penta_axis_ht.cpp or https://github.com/DerAndere1/Marlin/blob/Marlin2ForPipetBot/Marlin/src/module/penta_axis_trt.cpp.