MarlinFirmware/Marlin #26334: Automated Bed Leveling (Bilinear) is not performing correctly.

GenXEngineer, 2023-10-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

 ABL - Bilinear matrix is being created and stored to EEPROM without trouble, (verified M503) but when the correction is applied to a print (either G29 or M420 S1) there is a clear bias towards the right hand side of the build plate. More specifically the right hand side of my 400mm build plate has about a 0.2mm compression that the left hand side does not have. It's very much like the entire correction is slighly tilted toward the right hand side.  I have discovered this error on multiple printers over the last two years. The larger format machine I just built makes it intolerable, however.  I'm a mechanical engineer with 25 years experience BTW inclulding CMM.  Just would like to see this solved for the community and of course myself. There's a little bug hiding in there somewhere. I realized getting good feedback data might be tough given all the variables and skill levels involved. Also, when using VSCode to compile my custom Marlin setup I do see a small littany of compiler errors about incorrect integer conversion for the bilinear leveling portion (sorry I forget the exact error nomenclation but can get that if you need). Those are the only soft warning errors I see during compile (maybe not significant I'm no programmer :).  I'm using a CR touch bed probe that the probe repeatibilty test states has a standard deviaion of only 0.003mm.     

### Bug Timeline

Old issue been going on for a couple years now

### Expected behavior

My machines are all mechanically sound, plumb, square tight etc. Also using a fixed 4mm thick borosilicate glass plate printer bed that is very flat (within 0.08mm) and trammed to near perfection. I use ABL to ensure that any bumps or wear over time don't allow the nozzle to impact the glass plate. Would expect that ABL would yield the same or better results than the glass plates alone.  I allow the bed to fully warm up before leveling. Also when trammed to perfection ABL off prints go down just fine. I have tried leveling correction limit set from 10 to 100mm and have tried 15x15 and 4x4 matrix sizes. All give same error.  I see other reports of same issue online going back to 2020.

### Actual behavior

It took way too many hours of grief to finally corner this one, but like I stated above the software correction appears to have an error. The nozzle nearly hits glass on the right side (0.2mm layer) when the left hand side is ok. It gradually tapers closer and closer from left to right just like the entire correction matrix is tilted.  It finally gelled late last night (I was peeved) when I placed a 3mm thick shim under only the left hand side of the bed to raise it up (a ridiculous amount). I just wanted to see how good ABL was tracking. I was surprised that even with that extreme amount of a glass bed tilt the ABL correction printed my test pattern just fine - but stiiiiiiiiiiiillll with the right hand side being compressed to the 0.2mm amount. Looked exactly the same as the trammed print. Kinda impressive if only it wasnt wrong.... So that told me right there and then that this is a software issue.   

### Steps to Reproduce

Get a printer that is well setup 
Heat and tram to perfection a glass bed that is flat (dont even need ABL)
Run & Store ABL
Run test print using ABL
Place flags on dual z motor rods to ensure active fixing is actually going on.
Print a large square perimeter pattern (Cura) The G Code is perfectly fine. 
Print same pattern with ABL off
See the problem

### Version of Marlin Firmware

10/8/23 nightly bug fix - but also going back to 2.0

### Printer model

Custom Ender 3 Pro 

### Electronics

Big Tree Tech SKR mini e3v3 via sd card

### Add-ons

CR touch proble and dual z motors and rods (all new)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm more than happy to help test fixes if you find anything and can provide screen shots of the compiler errors.

Thanks.

EvilGremlin, 2023-10-09:
You describe typical result of using probe on cantilevered gantry with large X offset and/or hotend getting tensioned by bowden tube. There was a bug with mesh apllied mirrored at some point. 
Did you check UBL? Do so. And you did not attached configs.

GenXEngineer, 2023-10-09:
> You describe typical result of using probe on cantilevered gantry with large X offset and/or hotend getting tensioned by bowden tube. There was a bug with mesh apllied mirrored at some point. Did you check UBL? Do so. And you did not attached configs.

Thanks for the response EG.  Just to add some more information to the conversation, the print head has been upgraded to the Sprite Pro direct drive and the filament is free fed to the sprite from the middle of the machine (above of course).  Thought of trying UBL but thought maybe it might use the same ABL logic? Will add configs.

GenXEngineer, 2023-10-09:
Here is a picture of the compiler errors FWIW...
[image]

GenXEngineer, 2023-10-09:
And here is a picture of an ABL applied print - note the left side is high and the right side is low (by about 0.2mm)
In the foreground you can see the line tapering fatter to the right. And Linear Advance rocks BTW and is applied here.

[image]

GenXEngineer, 2023-10-09:
And finally here is a picture of ABL applied but probed with a sheet of paper held under the right two most probe points (4x4 matrix). Or you can tilt the glass sheet by lifting the right side up by 0.2mm while probing - then pull it out as the print starts. Either way it turns out perfect. This is also how it turns out with ABL turned off as the glass plate is trammed perfectly. The lines are laid down about equal.

[image]

GenXEngineer, 2023-10-09:
Same behavior on all 5 printers.

[image]

mjfsch, 2023-10-10:
I cant really add anything technical here, but this seems to ring very familiar with my printer so i can add report of a similar experience.

I seem to always have a difference in level at the rear right quadrant of the bed, this is using a 3D touch sensor, and glass bed. 
it seems to be the same regardless of bilinear or UBL levelling.

it doesn't seem to matter how the bed is adjusted, the issue always appears to be consistent, the first layer prints fine anywhere else on the bed, but as it approaches the rear right the extrusion gets visibly more "squished" in the most extreme rear right almost to the nozzle been in contact with the bed and when moving out of this area there is an evident over extrusion as pressure is released.

I'm currently running Marlin 2.1.2 with UBL enabled, Z Fade disabled. but have previously run a number of different builds including bug fix builds and have always as far as I can remember had this issue.

Printer a bed slinger 250x250 bed with liner rails, dual Z axis, using an octopus pro board.

GenXEngineer, 2023-10-10:
Ok, have some more data to provide. Performed a statistical analysis on probe accuracy across the bed from the ABL routine. I ran 5 consecutive G29 V4 routines from PronterFace and obtained the following matrices.  The average standard deviation from each position measured over 5 runs is 0.004mm. Which is very close to the Z probe accuracy  test results. Fantastic. In my mind this once again rules out anything other than the firmware (or my firmware setup) causing the right hand side compression problem. I have verified that the ABL data is not mirrored or anything like that. 
[PRINTER ABL STD DEV.xlsx](https://github.com/MarlinFirmware/Marlin/files/12861884/PRINTER.ABL.STD.DEV.xlsx)
 I'll upload the excel file with the data for review. Thanks!

[image]

thinkyhead, 2023-10-11:
ABL is mathematically sound, but it relies on the probe to be accurate and consistent across the whole range of motion. "[One side high, one side low](https://marlinfw.org/docs/basics/troubleshooting.html#one-side-high,-one-side-low)" is a common symptom of a probe that is slightly changing its angle to the bed across some range of motion. There are several mechanical things that can lead to this issue:
- Twisted X Gantry
- Wires pulling on the Carriage
- Over/under-tightened eccentric X wheel.

You can try the `MECHANICAL_GANTRY_CALIBRATION` feature, which was created to help mitigate this issue, which commonly afflicts this particular style of printer.

ellensp, 2023-10-11:
wtf?
#define X_HOME_DIR -2
#define Y_HOME_DIR -2
#define Z_HOME_DIR -2

GenXEngineer, 2023-10-11:
> There are not valid
> 
> #define X_HOME_DIR -2 #define Y_HOME_DIR -2 #define Z_HOME_DIR -2
> 
> valid options are -1 or 1

Will fix indeed - thanks! Strangely it works fine, must default back to the correct valule. I liked your original response better BTW - although it did seem beneath your normal level of response. lol.

GenXEngineer, 2023-10-11:
> ABL is mathematically sound, but it relies on the probe to be accurate and consistent across the whole range of motion. "[One side high, one side [...]

That does make sense, I can see where given the offsets a rotation about the x axis woudl cause the probe to lift or drop more than the nozzle does. I'll perform a check later today and provide pictures. Will mount a paper clip (as near z probe location as possible) that hits the bed at the same exact time the nozzle does. Then we will skate around the bed and see if that relationship is maintained. If it is.... I'll do the math on how much twist would be required to life 0.2mm but wow that would take alot. Good stuff guys and gals thanks.

thisiskeithb, 2023-10-11:
Check out `X_AXIS_TWIST_COMPENSATION` as well:

https://github.com/MarlinFirmware/Marlin/blob/6f68da5be4dcc56cc6577f3727f0e240e4086785/Marlin/Configuration_adv.h#L1502-L1514

[`M423` - X Twist Compensation docs](https://marlinfw.org/docs/gcode/M423.html)

GenXEngineer, 2023-10-11:
Alllllllllrrrright. So please forgive the crudity of the model, but what I show here is my interpretation of what you are saying. According to calculations (with my setup) it would require a twist of only about 0.4 degrees over 400mm of x travel to obtain a 0.25mm compression.  Give or take a little. So for fun here are the illustrations.

1.) Model show at 5 degrees twist for visualization. The print head and x bar are all to scale and real dimensions.

[image]
[image]

2.) Model shown at 0.4 degrees twist for visualization. This might not be noticeable to the nake eye.

[image]
[image]

Will try to recreate this later on the actual printer to verify.

Best - GXE.

GenXEngineer, 2023-10-11:
> Check out `X_AXIS_TWIST_COMPENSATION` as well:
> 
> https://github.com/MarlinFirmware/Marlin/blob/6f68da5be4dcc56cc6577f3727f0e240e4086785/Marlin/Configuration_adv.h#L1502-L1514
> 
> [`M423` - X Twist Compensation docs](https://marlinfw.org/docs/gcode/M423.html)

This is fantastic, hope it works - thanks!

stef-evotech, 2023-10-12:
Hi GXE,

I think my issue was mentioned somewhere in this thread, this was a bug triggered by fade height settings, so if you want to read up:

https://github.com/MarlinFirmware/Marlin/issues/24002

Just food for thought:

Other than that, systematic lean to one side of the bed, might be to probe heat up while the bed is being measured. We observed this with inductive bed probes but maybe your touch probe is influenced by heat as well. For a quick test you could move the head of your printer to another position before starting the probing manually. Marlin would then start with the closest measuring point, so you could change the sequence of the probing and check if anything changes! Perhaps you could report your measurments via "G29 T"?

GenXEngineer, 2023-10-12:
Thanks for the help everybody - great group!!! You have earned and will receive a nice donation from the GXE. The problem was an X axis twist after all. All the times I've read thought Adv.h I never noticed that routine. Anyhow, here's pictures of what I've been chasing after for 3 years and hundreds of prints. Now let's just hope this isn't a routine (M423) that I have to run very often PITA - hopefully not roller induced !!! Peace all.

1.) The faux fixed probe try - first check to see if there was a problem (confirmed):

[image]
[image]
 
2.) After M423 and G29 - Printed perfect first layer border:

[image]
[image]
[image]

@ellensp @thinkyhead @tombrazier @thisiskeithb

closed 2023-10-12 by GenXEngineer