MarlinFirmware/Marlin #26336: [BUG] "always_inline function might not be inlinable": warning / [FR] `STM32F103RC_creality_maple` added to Creality_V4+

classicrocker883, 2023-10-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

since the recent commit 🩹 Remove extraneous 'inline'
See https://github.com/MarlinFirmware/Marlin/issues/26085

I receive this warning using the **stmf1-maple.ini**
`env: STM32F103RE_creality_maple`

```r
Marlin\src\HAL\STM32F1\MarlinSerial.cpp:31:29: warning: always_inline function might not be inlinable [-Wattributes]
 static __always_inline void my_usart_irq(ring_buffer *rb, ring_buffer *wb, usart_reg_map *regs, MSerialT &serial) {
```

side note (Feature Request):
`env: STM32F103RC_creality_maple` should be added to the CREALITY_V4...422/427 in **src/pins.h** file

### Bug Timeline

new, since  🩹 Remove extraneous 'inline'

### Expected behavior

I do not expect a warning to appear.

### Actual behavior

warning when focing "inline" removed

### Steps to Reproduce

1. Use a standard Ender-3 V2 configuration
2. Set to `STM32F103RE_creality_maple` as default_envs in platformio.ini
3. Compile

[Ender3V2Configs.zip](https://github.com/MarlinFirmware/Marlin/files/12878084/Ender3V2Configs.zip)

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-10-12:
I did a quick test using `STM32F103RC_btt_maple` (which according to #26085 received a fatal error) and such error did not occur for me.

having said, I do not know if it did use `toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)`, I didn't mess with changing that sort of thing. So I think we can revert that commit and put back `inline`.

ellensp, 2023-10-12:
maple is obsolete

going back to it is so wrong on so many levels.

It is not the correct way to support GD32

closed 2024-01-26 by github-actions[bot]

classicrocker883, 2024-01-29:
> maple is obsolete
> 
> going back to it is so wrong on so many levels.
> 
> It is not the correct way to support GD32

just wanted to mention that I heard it might be possible to update the bootloader to accept `STM32F103RC_creality` without the maple. but that requires something like St-link.