MarlinFirmware/Marlin #26337: [FR] (big machine, 2000mm by 2000mm by 2000mm axis movements - menus)

mmatus1112, 2023-10-12:
### Is your feature request related to a problem? Please describe.

I have a machine that is 2000 mm x 2000 mm x 2000 mm. In move menu, I am able to move the axes by 10mm, but due to the increased dimension of the machine, would it be possible to modify the code to be able to move it by 100 mm and 500 mm.

Also, when I move more than 999 mm, the display does not show the 1000 mm. Is it possible to change the display to show 4 digits instead of 3?

### Describe the feature you want

I want the movement menu to allow movements greater than 10 mm. For instance, I would love my machine to move by 100mm and 500 mm. Also, I want the movement display to show 4 digits instead of 3 (example, show 1000.0 mm).

### Additional context

My machine is 2000 mm by 2000 mm by 2000 mm, therefore big movements are required to control the device.

ellensp, 2023-10-14:
Marlin supports several display types, each of which has its own set of features, what display are you talking about?

For example the stock 128x64 mono display already has some of what your requesting

If you search the code for HAS_LARGE_AREA

you find 
```cpp
#define HAS_LARGE_AREA (TERN0(HAS_X_AXIS, (X_BED_SIZE) >= 1000) || TERN0(HAS_Y_AXIS, (Y_BED_SIZE) >= 1000) || TERN0(HAS_Z_AXIS, (Z_MAX_POS) >= 1000))
#if ENABLED(LARGE_MOVE_ITEMS)
  #define HAS_LARGE_MOVES true
#elif ENABLED(SLIM_LCD_MENUS)
  #define HAS_LARGE_MOVES false
#else
  #define HAS_LARGE_MOVES HAS_LARGE_AREA
#endif

``` 

As long as you don't over ride it by selecting SLIM_LCD_MENUS, when any of the build dimensions are >= 1000  the HAS_LARGE_MOVES flag is set

If you then search for HAS_LARGE_MOVES

You find in Marlin/src/lcd/menu/menu_motion.cpp that marlin automatically adds 50mm and 100mm move menu items,

```CPP
  else {
    if (HAS_LARGE_MOVES) {
      SUBMENU(MSG_MOVE_100MM, []{ _goto_manual_move(100); });
      SUBMENU(MSG_MOVE_50MM, []{ _goto_manual_move(50); });
    }

```

You can simply add the line 
      SUBMENU(MSG_MOVE_1000MM, []{ _goto_manual_move(1000); });
and add the language file string MSG_MOVE_1000MM  to Marlin/src/lcd/language/language_en.h under the MSG_MOVE_100MM line

X and Y also display a 4 digit position without issues.  But Z is limited to 3 digits

eg

[image]

Z can be easily updated to support this by repacing

    TERN_(HAS_Z_AXIS, strcpy(zstring, is_inch ? ftostr42_52(LINEAR_UNIT(lpos.z)) : ftostr52sp(lpos.z)));

with

    TERN_(HAS_Z_AXIS, strcpy(zstring, is_inch ? ftostr42_52(LINEAR_UNIT(lpos.z)) : ftostr4sign(lpos.z)));

vlsi, 2023-10-14:
I wonder if this issue is a duplicate of https://github.com/MarlinFirmware/Marlin/issues/26262

ellensp, 2023-10-15:
Feature added.

closed 2023-10-15 by ellensp

mmatus1112, 2023-10-16:
Thank you very much, Ellensp!

So I was able to add everything, except for the first part. So, I am using Marlin 2.0.8.3, and I believe for this version the HAS_LARGE_MOVES part was not yet developed. You may ask why am I not using the newer version of Marlyn, and it is because I get an error when I define 0 Extruders. 

So, I am using Marlin to control an automated frame that moves in the X, Y and Z direction. The frame is installed inside a wind tunnel. The rake moves 8ft by 8 ft by 6 ft (width, depth and height). I use the frame to install a probe to measure wind speeds at different locations (x,y and z). Now, I am using a RAMPS 1.4 board and it brings the following ports: X1, Y1, Z1, E0 and E1. When not using the E0 and E1, they become X2 and Y2.

The new Marlin gives me an error when I set extruders to 0 and therefore I am forced to use the version 2.0.8.3.

Sorry for my lack of knowledge to do this on my own :(

Thank you very much, Ellen!!!!!

P.S. I did post this same issue years ago, but it seems the post was deleted as I was not able to find it. I believe it was you who guided me to do all the necessary changes...

mmatus1112, 2023-10-16:
ok, so I was able to add the four digits to the X, Y and Z main screen.
[image]
[image]

Now, I cannot make the movement screen to show 4 digits:
[image]

thanks for your help!!!!

ellensp, 2023-10-16:
Line 88     in Marlin/src/lcd/menu/menu_motion.cpp 
  MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? ftostr41sign(pos) : ftostr63(pos));

change it to 

   MenuEditItemBase::draw_edit_screen(name, ui.manual_move.menu_scale >= 0.1f ? (HAS_LARGE_AREA ? ftostr51sign(pos) : ftostr41sign(pos)) : ftostr63(pos));

ellensp, 2023-10-16:
there aren't any known issues with disabling E on bugfix, but on removing an axis, you need to edit  all setting that has a E element also.

eg #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 500 }

This has X,Y,Z and E elements

would need to become  #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400 }

And there are about 15 of these sort of defines that you need to update

mmatus1112, 2023-10-19:
Thank you! After you mentioned I had to change all the other items, it worked with the latest version of Marlin! Thank you so much for he help!