MarlinFirmware/Marlin #26338: Upload method for DWIN Display[FR] (feature summary)

cyberdevnet, 2023-10-12:
### Is your feature request related to a problem? Please describe.

no related to a problem

### Are you looking for hardware support?

Dwin display

### Describe the feature you want

Have someone already implemented or found a method to upload the DWIN_SET files to DWIN Display other then using the SD Card?
What I would like to achieve:
- Upload the DWIN_SET folder to the MCU Flash (ESP32 in my case)
- Download the files to the Display using the Serial connection

Before I start digging into the code, I would like to know if someone has already done something similar.

Thanks!

ellensp, 2023-10-12:
This is just not possible

The DWIN display is its own little operating system, it doesn't allow for such things

It is closed source

Also note 75% of the dwins used (e2v2 type with encoders) are emulated, they use same protocol but everything else is undocumented.

These displays in general have lots of flash and very little ram, so you cannot upload all graphics assets at runtime. 

.

cyberdevnet, 2023-10-12:
one can upload the files with the tool they provide, or even using this .NET and Powershell https://github.com/UdiAzulay/DgusDude, wouldn't be possible to the same with a microcontroller?

The-EG, 2023-10-12:
According to the DWIN docs (here's a google translated one for reference):
[T5UIC1.Kernel.Application.Guide_V23.-.google.translate.pdf](https://github.com/MarlinFirmware/Marlin/files/12887331/T5UIC1.Kernel.Application.Guide_V23.-.google.translate.pdf)
It would supposedly be possible to do this (commands 0x31 for write and 0x32 for read).

However, if that would actually works remains an unanswered and very valid question. Even on my genuine DWIN, I found that it didn't support all the functions documented. And as ellensp, most screens shipped as "DWIN" aren't even DWIN and all the clones just emulate the same protocol, but with even less functionality. At least one doesn't even support screen rotation.

So, even if you were able to do this on a genuine DWIN, it wouldn't work for what most users are receiving with new printers now.

The-EG, 2023-10-12:
> https://github.com/UdiAzulay/DgusDude, wouldn't be possible to the same with a microcontroller?

I'm not sure this is even the same controller. That does list T5, but the genuine DWIN in the "Ender 3 v2 screen" is a T5UIC1.

The serial protocol and features vary widely between models.

ThomasToka, 2023-10-12:
it is only possible with a usb dongle that is connected to the display with a cable. that is what dgus guy makes. you can send files and interact with the display directly. send commands and get files from the device. over the serial connection this is not possible afaik.

ThomasToka, 2023-10-12:
> According to the DWIN docs (here's a google translated one for reference): [T5UIC1.Kernel.Application.Guide_V23.-.google.translate.pdf](https://github.com/MarlinFirmware/Marlin/files/12887331/T5UIC1.Kernel.Application.Guide_V23.-.google.translate.pdf) It would supposedly be possible to do this (commands 0x31 for write and 0x32 for read).
> 
> However, if that would actually works remains an unanswered and very valid question. Even on my genuine DWIN, I found that it didn't support all the functions documented. And as ellensp mentioned, most screens shipped as "DWIN" aren't even DWIN and all the clones just emulate the same protocol, but with even less functionality. At least one doesn't even support screen rotation.
> 
> So, even if you were able to do this on a genuine DWIN, it wouldn't work for what most users are receiving with new printers now.

screen rotation in most cases is no problem. it is set with byte 0x05. you even can sent it from an toggle. depending on the display the byte is 0 or 39 or 62..

The-EG, 2023-10-12:
> screen rotation in most cases is no problem. it is set with byte 0x05. you even can sent it from a toggle. depending on the display the byte is 0 or 39 or 62..

Not relevant to this, nor does that change the fact that the 'DACAI' displays have been verified to completely ignore that instruction all together.

ThomasToka, 2023-10-12:
The dacais  can do everything that the dwin can. the difference is that the dacais internal firmware is not open source. or better said not all functions are unlocked by the vendors. I can tell you for example that creality released with the F28 Firmware a new firmware.zlib that for example unlocks gcode preview functionality (and i mean inside the display so that it is possible to send a thumbnail to it with marlin) and the sound on off toggle for the dacais. and yes i can rotate my dwin and dacais with a toggle if i want. and yes i know that not all act the same.

back to topic: still not possible without a dwin debugger that is connected to the display. look up youtube an marias or savannahs videos.

classicrocker883, 2023-10-16:
> one can upload the files with the tool they provide, or even using this .NET and Powershell https://github.com/UdiAzulay/DgusDude, wouldn't be possible to the same with a microcontroller?

from the README:

>About: I originally developed it after failing to upload a custom firmware to my 3d printer using the SD card, for some reason my SD interface stop loading the update screen and it seems that my version of DWIN OS is NoAck (does not response 0x4F4B for 0x83 commands) so for me, DWIN uploader natural tool fail with timeout error

may be a bit off topic but I believe this is important nonetheless. I just want to point out at how some (unfortunate) things happen for a reason. I mean just look at this example. God so happen to let this person receive a defective item, who has the **know-how** to make it work (in his case) which can help others. Imagine if this unfortunate broken thing didnt happen - if he received a working product. or what if this happened to someone else, they would have NO idea probably. then we wouldnt have this kind of "fix".

basically if you look at past events, like the Titanic sinking, yeah how unfortunate, and the lives lost. However, on the bright side, that event did improve safety overall. Same with how fire-escape ladders came about. Think of the sacrifice which in turn saved countless lives and improved the lives of others.

so next time you get stuck at a red light, or someone cuts you off, or whatever makes your day worse, instead of getting
mad and frustrated, think things are supposed to happen.

kenc106, 2025-02-23:
This is an old post but wanted to suggest this solution, its quiet elegant:

change the mini SD card to full side or use adapter.
Buy a Toshiba flashair SD Card
permently instal the flashair card in the SD Slot
the monitor won;t boot with data on card, populagte sd card over the air, reboot to import changes, delete file on sd card and then reboot the monitor via power off/on

this should allow updating over the air.

closed 2026-05-18 by thinkyhead