MarlinFirmware/Marlin #26357: [BUG] GCODE_REPEAT_MARKERS break POWERLOSS RECOVERY DEBUG

ThomasToka, 2023-10-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Enabling 
`#define GCODE_REPEAT_MARKERS `
 and
`#define DEBUG_POWER_LOSS_RECOVERY`

breaks compilation.

### Bug Timeline

known for some months

### Expected behavior

No compilation error:

### Actual behavior

```
In file included from Marlin\src\feature\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\feature\powerloss.cpp:27:
Marlin\src\feature\powerloss.cpp: In static member function 'static void PrintJobRecovery::debug(FSTR_P)':
Marlin\src\feature\powerloss.cpp:586:64: error: 'uint8_t Repeat::index' is private within this context
           DEBUG_ECHOLNPGM("repeat index: ", info.stored_repeat.index);
                                                                ^
Marlin\src\feature\../inc/../core/macros.h:640:24: note: in definition of macro 'EVAL1'
 #define EVAL1(V...)    V
                        ^
Marlin\src\feature\../inc/../core/macros.h:639:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
Marlin\src\feature\../inc/../core/macros.h:638:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))
 ...
 ```

### Steps to Reproduce

Enable:
#define SDSUPPORT
#define POWER_LOSS_RECOVERY
#define GCODE_REPEAT_MARKERS 
#define DEBUG_POWER_LOSS_RECOVERY

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

does not matter tested with mega2560

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configs not needed. Happens with every that has the both enabled.

ThomasToka, 2023-10-22:
@ellensp should i add this to my pr i just opened?

ellensp, 2023-10-23:
no it need done properly..  this is just a hack to get past the error

ellensp, 2023-10-23:
nah lets keep them separate, since you pr is complicated enough.

Created a PR to fix this.

closed 2023-10-23 by ellensp