MarlinFirmware/Marlin #26358: [BUG] (bug summary)

Puch-tdi, 2023-10-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Vertex K8400 cooling fan does not work on non of the Marlin V2.x.x versions.
The pins_K8400.h have no configuration for the FAN_PIN.
First i thought my board was defective so i reprogrammed with Marlin V1.4 and it worked again.
After V2.1.1 and 2.1.2.1 trials still not working so i went inspection some files.
If you follow through the files it points to pins_3DRAG.h wich point again to pins_RAMPS.h
There i saw the FAN_PIN is 4 (wich could be correct for a RAMPS board, i don't know for sure) but the K8400 uses pin 8 for the fan.
So i went peeking in the pins_K8600.h and pins_K8800.h files and saw they have a seperate rule wich point to pin 8 for the cooling fan.
After writing this rule to the pins_K8400.h file it finally works correctly.

### Bug Timeline

On all V2.x.x marlin versions

### Expected behavior

the cooling fan should be started

### Actual behavior

nothing happend

### Steps to Reproduce

writing the correct pins file

### Version of Marlin Firmware

2.1.1

### Printer model

Vertex K8400

### Electronics

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I think it would be clear, i only adjusted pins_K8400.h file for this.
This is the original file with one minor adjustment to get it working under the "Heaters / Fans" section
[pins_K8400.zip](https://github.com/MarlinFirmware/Marlin/files/13060539/pins_K8400.zip)

ellensp, 2023-10-21:
FAN_PIN no longer exists. replaced with FAN0_PIN  in bugfix

Please attach config files, as requested

ellensp, 2023-10-21:
There are two example configs for this machine [Dual-head](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Velleman/K8400/Dual-head) and [Single-head](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Velleman/K8400/Single-head)

Single head 
pins_3DRAG.h  sets: 
#define MOSFET_B_PIN                           8

pins_RAMPS.h  sets:
#define FAN0_PIN                MOSFET_B_PIN

Dual-head

pins_3DRAG.h  sets: 
#define MOSFET_B_PIN                           8
#define MOSFET_C_PIN                           9

pins_RAMPS.h  sets: 
#define HEATER_1_PIN       MOSFET_B_PIN
#define FAN0_PIN                MOSFET_C_PIN

Then pins_K8400.h has:
#undef HEATER_1_PIN
#define HEATER_1_PIN                          11

So Single head  fan0 is on 8 MOSFET_B_PIN  
Dual head fan0 ends up on 9 MOSFET_C_PIN

So is t dual head configuration your referring to?

Puch-tdi, 2023-10-21:
Sorry, forgot to mention that.
I do have dual heads, but the mainbord only have one output for the cooling fan and one output for heatbreak fan.
I did build my own casing but on the original K8400 i onces bought the fans were allready wired parallel for the use of extra fans

Puch-tdi, 2023-10-21:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/13060791/config.zip)
Hereby the config and config adv files

ellensp, 2023-10-21:
I agree this should be added, created a pr to fix it.

closed 2023-10-21 by ellensp