MarlinFirmware/Marlin #26360: [BUG] missing ftpl in marlinui_TFTGLD.cpp

narno2202, 2023-10-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

since last Marlinui menu  tweaks merge ( #27068) , CI fails  in LPC1769 (first failure in tests)  with `marlin/src/lcd/TFTGLCD/marlinui_TFTGLCD.cpp:978:25: error: 'fstr' was not declared in this scope`.
There are some `ftpl` missing in `void MenuItem_static::draw(const uint8_t row, FSTR_P const ftpl, const uint8_t style/*=SS_DEFAULT*/, const char *vstr/*=nullptr*/)`. And corrected below :

```
void MenuItem_static::draw(const uint8_t row, FSTR_P const ftpl, const uint8_t style/*=SS_DEFAULT*/, const char *vstr/*=nullptr*/) {
    if (!PanelDetected) return;
    lcd_moveto(0, row);

    uint8_t n = LCD_WIDTH;
    const bool center = bool(style & SS_CENTER), full = bool(style & SS_FULL);
    const int8_t plen = ftpl ? utf8_strlen(ftpl) : 0,
                 vlen = vstr ? utf8_strlen(vstr) : 0;
    int8_t pad = (center || full) ? n - plen - vlen : 0;

    // SS_CENTER: Pad with half of the unused space first
    if (center) for (int8_t lpad = pad / 2; lpad > 0; --lpad) { lcd.write(' '); n--; }

    // Draw as much of the label as fits
    if (plen) n -= lcd_put_u8str(ftpl, itemIndex, itemStringC, itemStringF, n - vlen);

    if (vlen && n > 0) {
      // SS_FULL: Pad with enough space to justify the value
      if (full && !center) {
        // Move the leading colon from the value to the label
        if (*vstr == ':') { lcd.write(':'); vstr++; n--; }
        // Move spaces to the padding
        while (*vstr == ' ') { vstr++; pad++; }
        // Pad in-between
        for (; pad > 0; --pad) { lcd.write(' '); n--; }
      }
      n -= lcd_put_u8str_max(vstr, n);
    }

    for (; n > 0; --n) lcd.write(' ');

    lcd.print_line();
 }
```

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-10-21:
Yes, just added pr to fix this https://github.com/MarlinFirmware/Marlin/pull/26359

closed 2023-10-21 by ellensp

thisiskeithb, 2023-10-21:
This will actually be fixed in https://github.com/MarlinFirmware/Marlin/pull/26339