MarlinFirmware/Marlin #26364: [FR] (feature summary)

j007jackson, 2023-10-22:
### Is your feature request related to a problem? Please describe.

Yes. I am currently having an intermittent problem with my Ender 3 Pro, running Marlin 2.1.2.1.
On long prints (multiple hours) the printer stops and reboots. 
This is a difficult issue to diagnose, because of the long print time time and lack of any surviving error or warning messages.
I would love to be able to see the the log message that was issued just before shutdown and restart.
Thus, my request for a new feature that would me let me see the last log message.
I know the issue of SD card logging has been discussed before, but this is a limited special case version, just for cases like mine.

### Describe the feature you want

When the firmware gets in a situation where it has determined that a shutdown is required.
It would first close the gcode file and do whatever clean up actions are required.
It would save the last log message and then write that message to a special file on the SD card
(Marlin.log perhaps) and then proceed with the shutdown. You could overwrite any previous contents, just to prevent the file from expanding beyond control.
This would help greatly in identifying the problem.

EvilGremlin, 2023-10-22:
use serial logging in octoprint

ellensp, 2023-10-23:
Please see https://marlinfw.org/docs/gcode/M928.html
Pair that with POSTMORTEM_DEBUGGING and you'll have all kinds of debug info