MarlinFirmware/Marlin #26371: MarlinConfigPre.h: No such file or directory

menthok, 2023-10-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

 *  Executing task: C:\Users\s3dcn\.platformio\penv\Scripts\platformio.exe run 

Processing esp32 (platform: espressif32@2.1.0; board: esp32dev; framework: arduino)
------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, 
olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10004.210126 (1.0.4)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP3DLib @ 1.0.12+sha.cdeecd2
|-- U8glib-HAL @ 0.5.2
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3
|-- WebSockets @ 2.3.4
|-- ESP32SSDP @ 1.1.1
|-- SPI @ 1.0
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|-- ESPmDNS @ 1.0
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- SPIFFS @ 1.0
|-- Wire @ 1.0.1
Building in release mode
Compiling .pio\build\esp32\src\src\HAL\ESP32\FlushableHardwareSerial.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\HAL.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\HAL_SPI.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\Servo.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\Tone.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\WebSocketSerial.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\eeprom.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\i2s.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\ota.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\spiffs.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\timers.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\u8g_esp32_spi.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\web.cpp.o
Compiling .pio\build\esp32\src\src\HAL\ESP32\wifi.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\esp32\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\esp32\src\src\MarlinCore.cpp.o
Compiling .pio\build\esp32\src\src\core\serial.cpp.o
Compiling .pio\build\esp32\src\src\core\utility.cpp.o
Compiling .pio\build\esp32\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\esp32\src\src\gcode\config\M309.cpp.o
[... 296 lines cut ...]
esp32          FAILED    00:10:49.289
================ 1 failed, 0 succeeded in 00:10:49.289 ================ 

 *  The terminal process "C:\Users\s3dcn\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

### Version of Marlin Firmware

Marlin 2.1.2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-10-26:
Probably the same as https://github.com/MarlinFirmware/Marlin/issues/25767

menthok, 2023-10-26:
> 

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02010201

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

// @section info

// Author info of this build printed to the host during boot and M115
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
//#define CUSTOM_VERSION_FILE Version.h // Path from the root directory (no quotes)

/**
 * *** VENDORS PLEASE READ ***
 *
[... 3269 lines cut ...]

// @section servos

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands

// (ms) Delay before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

// Disable servo with M282 to reduce power consumption, noise, and heat when not in use
//#define SERVO_DETACH_GCODE

closed 2024-09-23 by github-actions[bot]