MarlinFirmware/Marlin #26372: [BUG] [To verify] pins_CREALITY_F401.h (CR4NS200141C13) might be wrong

EvilGremlin, 2023-10-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

User have CR4NS200141C13 and built BOARD_CREALITY_V24S1_301F4 and everything work fine, homing with bltouch. Which shouldn't be possible as servo_pin is different, among other things. 

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

BeastCrawler, 2023-10-28:
Hello, here are the Config files and the compiled .bin file for the firmware I'm using. I'm sorry that I don't yet have a picture of my motherboard. I will get a good one soon.
[Bin file.zip](https://github.com/MarlinFirmware/Marlin/files/13194803/Bin.file.zip)
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/13194797/Config.Files.zip)

closed 2024-03-26 by github-actions[bot]

mriscoc, 2024-04-23:
Only to note that there is another board with similar name but with other pin configuration: https://github.com/MarlinFirmware/Marlin/pull/27003