MarlinFirmware/Marlin #26377: [BUG] M28 O consider position trusted after disabling steppers

vovodroid, 2023-10-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Currently `M28 O` do not perform homing after `M84` disable steppers, and consider that axes are trusted, though they could be moved. I guess `M84` should 'untrust' the corresponding axis position.

### Bug Timeline

On the Big Bang

### Version of Marlin Firmware

2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-10-29:
M28 - Start SD write
I suspect you mean G28

ellensp, 2023-10-29:
What controller?
Does it have shared enable lines?
Does it have any enable  lines?

vovodroid, 2023-10-29:
Here is my printer https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/B1%20SE

It uses TMC2208 (actually TMC2225), but I thing that turning off motor should always untrust axis position.

ellensp, 2023-10-29:
needed details to attempt to locate block of code responsible.

Shared enables is different code from individual enables.
if it does not have enable lines it cannot disable them...

Your controller, BOARD_BTT_SKR_V2_0_REV_B, has individual enable pins for each stepper driver.

vovodroid, 2023-10-29:
I don't really know about lines and so on. I suggest that M84 or timeout expiration just clear some "trusted" bits or something like that.

ellensp, 2023-10-29:
Confirmed issue is present in simulator

ellensp, 2023-10-29:
code has       ```if (TEST(b, a) && TERN(HOME_AFTER_DEACTIVATE, axis_is_trusted, axis_was_homed)(a))```
 
It is checking axis_was_homed, which is still true
axis_is_trusted is disabled as expected.

You require #define HOME_AFTER_DEACTIVATE to do what you want.

closed 2023-10-29 by ellensp

vovodroid, 2023-10-29:
Wouldn't it be reasonable to enable this by default?

thisiskeithb, 2023-10-29:
> Wouldn't it be reasonable to enable this by default?

No. `HOME_AFTER_DEACTIVATE` is annoying if you don’t want to home, so that’s why it’s an option.