MarlinFirmware/Marlin #26380: [FR] Prevent triggering of piezo and strain gauge probes during z-acceleration

Moeschus, 2023-10-30:
### Is your feature request related to a problem? Please describe.

Strain gauges and piezo sensors can be excellent probes. They detect the force of the nozzle hitting the build plate and then send a trigger signal. Unfortunately accelerations and vibrations also excert forces on these sensors and can lead to false trigger events. Vibrations caused by x and y accelerations are already taken care of by "DELAY_BEFORE_PROBING" in Configuration.h. This delays the start of the probing move after any x- or y-movement and thereby allows vibrations caused by these moves to settle. But then the start of the probing move itself causes inertial forces that even act in the sensor direction. These forces are even more likely to cause false triggers.

Just lowering the probing speed doesn't work for piezo sensors because they need a certain impact speed to trigger. Reducing the probing speed would lower the actual signal just as much as the noise caused by acceleration.

### Describe the feature you want

It would be great to have either 

- a possibility to limit the z-acceleration of probing moves

OR

- a possibility to ignore trigger signals for a certain timespan (or path length) after the start of a probing move. In this case z clearance values for probing moves would have to be checked or automatically adjusted to prevent hitting the print bed with disabled trigger inputs.

ellensp, 2023-10-30:
Also 

```
// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
```

should be disabled, so endstops are ignored except for homing moves

Moeschus, 2023-10-30:
Thank you very much for your fast response!

Unfortunately I currently can't enable ENDSTOP_NOISE_THRESHOLD to test it because I have sensorless homing on my x- and y-axis. And both are not compatible.

But would this even solve the problem? The noise (caused by z-acceleration) and the actual signal (when the nozzle hits the print bed) are similar in magnitude and length. Therefore I think the only way to distinguish between them would be timewise (-> disabling the trigger input in the beginning of the move). Another solution could be to lower the acceleration which would in turn lower the magnitude of the noise.

ENDSTOPS_ALWAYS_ON_DEFAULT is already disabled.