MarlinFirmware/Marlin #26382: MKS Robin pro defiition is missing pins for WIFI module

Lumicrro, 2023-10-31:
These lines need to be added to file "pin_MKSROBIN_PRO.h"

//
// MKS_WIFI pins
//
  #define WIFI_IO0_PIN                      PB11
  #define WIFI_IO1_PIN                      PC7
  #define WIFI_RESET_PIN                    PA5

ellensp, 2023-10-31:
are you sure about WIFI_IO0_PIN  ?  

PB11 is rxd0

Lumicrro, 2023-10-31:
Correct, it is not used.

But then the wifi init code needs to be checked as I get an error in "wifi_module.cpp" saying "WIFI_IO0_PIN" is not defined.

void mks_esp_wifi_init() {
  wifi_link_state = WIFI_NOT_CONFIG;
  SET_OUTPUT(WIFI_RESET_PIN);
  WIFI_SET();
  SET_OUTPUT(WIFI_IO1_PIN);
  SET_INPUT_PULLUP(WIFI_IO0_PIN);

Lumicrro, 2023-10-31:
I just did a test with 
#define WIFI_IO0_PIN -1

and it works, Marlin compiles well and WIFI module is working.

thisiskeithb, 2024-03-06:
Fixed in #26823

closed 2024-03-06 by thisiskeithb