MarlinFirmware/Marlin #26385: [FR] Combined LINEAR and BILINEAR modes.

vovodroid, 2023-11-01:
### Describe the feature you want

While AUTO_BED_LEVELING_LINEAR can compensate bed tilt, and AUTO_BED_LEVELING_BILINEAR can compensate uneven bed, why not combine both of them and get tilted, but not distorted model:

[image]

EvilGremlin, 2023-11-02:
UBL doing exactly that for many years, please read docs first before openeing issues.

vovodroid, 2023-11-02:
If you are talking about G29 G, it's not clear that it combines both modes. Anyway UBL is much more complex for setup and usage than just to issue G29 commmand like for linear or bilinear.

EvilGremlin, 2023-11-02:
J, not G. Well documented, what exacty is unclear?
And no, UBL is not more complex. Anyway, this already exist in UBLso redundant features wont be added to bilinerar.

closed 2023-11-02 by ellensp