MarlinFirmware/Marlin #26386: Filament runout sensor turning off

tommylux, 2023-11-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I would like to re-open issue 25351 (I can't it's locked!).

My additions to the issue:
After M600 command is issued and filament has been re-loaded, there is only 'Purge More' or 'Filament Runout ON'.

Expecting to see 'Continue option', this appears briefly when you toggle 'Filament Runout' to OFF (which sets to off and continues. If you attempt to switch back on, it immediately goes into a M600 process again.

In my opinion, both 'Filament Runout' Toggle and 'Continue' commands are currently occupying line 2, 'Filament Runout' text is taking precedence over 'Continue' and both commands are run when hitting line 2

### Bug Timeline

Sort of both a new issue reporting but this happens in the older 2.0.x marlin firmware

### Expected behavior

I expect to see a Continue button option in the on screen menu.

### Actual behavior

As described above the Filament Runout sensor turns OFF

### Steps to Reproduce

Start a print and let the Filament runout sensor load the filament then the error or issue should show up in the menu screen that is displayed.

### Version of Marlin Firmware

2.1.2.1

### Printer model

A20T

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-11-02:
No config files provided
No Images of what you talking about provided
No details of what display your using provided
No details on how you have the runout sensor connected

Please provide all these details.

Issue https://github.com/MarlinFirmware/Marlin/issues/25351 was closed as it looking like a support question, not a bug. It still looks like a support issue not a bug...

tommylux, 2023-11-02:
> No config files provided No Images of what you talking about provided No details of what display your using provided No details on how you have the runout sensor connected
>
> Please provide all these [...]

Hopefully more info now provided in details above.

ellensp, 2023-11-02:
And your config files?  So we can attempt to reproduce this ??

tommylux, 2023-11-02:
> And your config files?  So we can attempt to reproduce this ??

Sure I will provide config files later. Perhaps a combination of multiple parameters are at play with regards to Mixing extruder or whatnot, though I wouldn't have thought s since there is nothing that can be configured to alter the behaviour of M600 in config.

tommylux, 2023-11-02:
Hopefully all infomation you require has now been provided.

My guess is something to do with either of the following files:
host_actions.cpp..
Case 1 -"Continue" / "Disable Runout" button. Perhaps these should be seporate.

Or maybe something to do with DRAW_Button position for Continue?

ellensp, 2023-11-02:
The provided Config files says BOARD_GT2560_V4_A20
Where as previous issue says you have BTT Octopus Pro 429

Please clarify

Also do you have a V1 or the V2 run out sensor ?

thomasluxton, 2023-11-02:
> The provided Config files says BOARD_GT2560_V4_A20 Where as previous issue says you have BTT Octopus Pro 429
> 
> Please clarify
> 
> Also do you have a V1 or the V2 run out sensor ?

I do not know what a BTT Octopus Pro 429. I have a GT2560_V4. The other ticket referenced is a different user having the same issue with Marlin.

thomasluxton, 2023-11-02:
I do not know if it is a v1 or v2 run out sensor, but I can tell you that there is example Geeetech A20T config files which I always reference if that helps.

ellensp, 2023-11-02:
V1 looks like 

[image]

While V2 looks like 

[image]

tommylux, 2023-11-02:
Like this..
[image]

ellensp, 2023-11-02:
Referencing the other issue has led me astray. 
Not a Octopus controller or a BTT run out sensor

tommylux, 2023-11-03:
> Referencing the other issue has lead me astray. 
> 
> Not a Octopus controller or a BTT run out sensor   
> 
> 

I believe it is an interface issue.. ie it happens regardless of the model of the runout sensor.

In fact, I don't think it has anything to do with the runout sensor. You can trigger the behaviour with M600 regardless of the runout sensor pin. 

I would suggest that we look at M600 behaviour in the first instance as I don't believe it has anything to do with run out sensor.

I've suggested looking at certain file, the 'case 1' combines 'continue' and 'runout sensor state', which both seem to take line 2 of the UI. I believe this is the right track to troubleshoot the issue.

tommylux, 2023-11-03:
> Referencing the other issue has lead me astray. 
> 
> Not a Octopus controller or a BTT run out sensor   
> 
> 

Likewise referencing free original issue was 100% the right thing to do. It demonstrates the same behaviour when M600 is triggered, and that other people have experienced the same issue.

We're meant to search for issues first to see if they are being worked on before logging new issues.

thinkyhead, 2023-11-05:
I took a quick look at the code and pushed a possible fix at #26394 so have a look at that and see if it fixes the issue for you.

commit dba613fa (2023-11-05): 🔨 Fix test of env['PROGNAME']

Fixes #26386
PR #26394 (2023-11-16): 🩹 Fix runout state in menu
Get the `runout.filament_ran_out` state outside the menu loop to hopefully fix #26386.
commit 7fe07dc4 (2023-11-16): 🩹 Fix runout state in menu (#26394)

Fixes #26386
closed 2023-11-05 by thinkyhead